rust-embedded/wg #361: [RISC-V] codegen seems not ideal

japaric, 2019-06-19:
Disclaimer: I don't know much of the RISC-V ISA so this may be just my poor
understanding.

Some things I observed while working with the [hifive1] board.

[hifive1]: https://github.com/japaric/hifive1

> NOTE: wherever the (compilation) target is not specified,
> `riscv32imac-unknown-none-elf` was used

## `jalr` instead of `jal`

All direct function calls seems to use some sort of relocatable form. Example
below:

``` rust
#[no_mangle]
unsafe fn _start() -> ! {
    foo();

    loop {}
}

#[inline(never)]
#[no_mangle]
unsafe fn foo() {
    asm!("" : : : "memory" : "volatile")
}
```

``` console
$ cargo objdump --bin app --release -- -C -d -no-show-raw-insn
0000000000011000 _start:
   11000:       addi    sp, sp, -16
   11002:       sw      ra, 12(sp)
   11004:       sw      s0, 8(sp)
   11006:       addi    s0, sp, 16
   11008:       auipc   ra, 0
   1100c:       jalr    ra, ra, 10
   11010:       j       0

0000000000011012 foo:
   11012:       addi    sp, sp, -16
   11014:       sw      ra, 12(sp)
   11016:       sw      s0, 8(sp)
   11018:       addi    s0, sp, 16
   1101a:       lw      s0, 8(sp)
   1101c:       lw      ra, 12(sp)
   1101e:       addi    sp, sp, 16
   11020:       ret
```

That `auipc`, `jalr` combination looks like relocatable code to me (not that I
know much about relocatable code). I was expecting to see `jal` function calls
of this form:

``` rust
global_asm!(r#"
  .global _start
_start:
  addi    sp, sp, -16
  sw      ra, 12(sp)
  sw      s0, 8(sp)
  addi    s0, sp, 16
  jal     foo
  j       0
"#);

#[inline(never)]
#[no_mangle]
[... 279 lines cut ...]

Registers are pushed in the prologue and then popped in the epilogue of the
function.

---

It would be good to check if these backend issues have been fixed in the latest
version of LLVM (using `llc`) because `rustc` is using a several months old
LLVM. If a recent commit shows the same issues then we may want to submit bug
reports to the LLVM project.

## Metadata

``` console
$ rustc -Vv
rustc 1.37.0-nightly (04a3dd8a8 2019-06-18)
binary: rustc
commit-hash: 04a3dd8a872633ca1e4c217d11f741cc35cb19a5
commit-date: 2019-06-18
host: x86_64-unknown-linux-gnu
release: 1.37.0-nightly
LLVM version: 8.0
```

cc @rust-embedded/riscv @Disasm

Disasm, 2019-06-20:
Oh, someday I'll have to become an LLVM developer. I'm 99% sure these strange things have nothing to do with Rust itself, but I need to check anyway.
Thanks for the observations!

Disasm, 2019-06-24:
## `atomic::compiler_fence` produces an instruction
> Even though the description of the function states that "compiler_fence does
not emit any machine code".

The description is wrong. This `compiler_fence` lowers to `atomic_fence(ordering, SingleThread)` and this `atomic_fence` can be lowered to any architecture-dependent code, not only to empty code.
LLVM lowers `atomic_fence(ordering, SingleThread)` to `CompilerBarrier` for AArch64 and ARM targets, but not for AVR, PowerPC, RISC-V and Spark. Of cause, this "SingleThread means CompilerBarrier" hack could be added for RISC-V too, but I do not see any adequate reason for this. Something is wrong on the Rust side.

## `intrinsics::abort` != `UNIMP`
Are there any guidelines on `abort` implementation? I think that `UNIMP` instruction can cause "Illegal instruction" exception and further stack overrun. Infinite loop inside `abort` seems like a better idea to me.

## `noreturn` `nounwind` & divergent functions
The same for clang: https://godbolt.org/z/YPa98Z
However, this register save code elimination is quite complex thing because you need to proof that it's useless. At the same time we do not see register restores because this code was removed during the dead code elimination process, which is easier.

## `jalr` instead of `jal`
I do not have any answer yet, trying to investigate...

Disasm, 2019-06-30:
## `jalr` instead of `jal`
Compiler generates `call <something>` instruction, which translates into `auipc`+`jalr` pair (The RISC-V Instruction Set Manual Volume I: User-Level ISA v2.2, p.110, table 20.2). That what you can see in `*.o` files because at this point it's impossible to determine whether you can call your function with `jal`. However, linker could translate this to `jal` (or even `j`) instruction if it's instructed to do so with [linker relaxation](https://www.sifive.com/blog/all-aboard-part-3-linker-relaxation-in-riscv-toolchain) attribute. By default, LLVM do not use this attribute for some reason, but it's possible to turn it on.

Unfortunately in Rust this doesn't work: the attribute starts to appear in object files, but rust-lld ignores it and leaves instructions as is. I need to figure out why.

lenary, 2019-08-01:
Hi! I'm a developer at lowrisc, working with Alex Bradbury on the RISC-V llvm backend. I've just started (yesterday!) looking at where we can improve the RISC-V Rust backend, in addition to my main LLVM work.

The LLVM backend only recently became non-experimental (around the time of the 9.0 branch, with releases imminent). With that, the aim is that we can compile most RISC-V programs in `rv{32,64}imacfd`, supporting the main ABIs. We know there are still lots of opportunities to produce "better" code. I'm happy to note that none of these issues make your program incorrect, they just make it use a slightly

Looking at these issues:
- `jalr` instead of `jal`. You are correct that this is an optimisation commonly performed by the linker, during linker relaxation. I'm not sure LLD does linker relaxation yet, nor am I sure there are any outstanding patches so it may do so. GNU ld will definitely perform this optimisation (the correct relocations should be being emitted).
- `intrinsics::abort` I think we haven't really addressed many custom/intrinsic lowerings in the RISC-V llvm backend yet. I shall take a closer look at this.
- `atomic::compiler_fence` I will take a look at how this is lowered. It may be a case of doing a different lowering in the RISC-V backend, but somehow this feels like a higher-level issue than that. I'm going to have to look at some more resources, and see if this should instead be a change somewhere in the target-independent part of LLVM.
- Stack usage in `noreturn`, `nounwind` functions: Right now it seems like the compiler is naive, and saves registers it doesn't need to. We can look at this not happening (subject to the ABI), but it's not a top priority.

I hope this clears up what's going on!

SUPERCILEX, 2023-05-04:
For anyone working on branch prediction, this was just recently fixed: https://github.com/rust-lang/rust/issues/109426. Enable like this: https://github.com/SUPERCILEX/hardcaml_riscv/commit/d1125a68db16579dd58f893b26f3c2db65158ab3#diff-f34d15ec63ba1edb3a7a25b6a3e3b7d3c1f02cbef0b66a6a73e8cf5c959a8576

rmsyn, 2024-08-17:
## `jalr` instead of `jal`

This has been resolved since this issue was opened: <https://godbolt.org/z/zqET14TEs>

## `noreturn` `nounwind` and divergent functions

This has been resolved since this issue was opened: <https://godbolt.org/z/5bsEss6M3>

## intrinsics::abort != UNIMP

This has been resolved since this issue was opened: <https://godbolt.org/z/qv7GndhEc>

## `atomic::compiler_fence` produces an instruction

This issue seems to still exist in the current stable and nightly compilers: <https://godbolt.org/z/ao3464vbe>

---

Out of the original issues noted, the only remaining issue seems to be `atomic::compiler_fence` and `atomic::fence` emitting code.

Not sure if these should actually lower into a `fence` and/or `fence.i` RISC-V instruction, or what the desired behavior is.

jamesmunns, 2024-08-18:
The compiler fence seems to disappear when `-C opt-level="..."` is passed. It's likely the compiler fence is making a function call but doing nothing, but once optimized the actual call is inlined and the side effect is removed.

BartMassey, 2024-08-18:
Everything sounds like it's working more-or-less as expected now. Anyone object to closing this issue?

adamgreig, 2024-08-27:
Thanks for triaging!

closed 2024-08-27 by adamgreig