rust-embedded/wg #362: Embedded WG Meeting, Jun 25th

therealprof, 2019-06-20:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-06-25,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Reminders / announcements
- [Issues nominated for discussion, if any](https://github.com/rust-embedded/wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

jamesmunns, 2019-06-23:
Discuss the outcome of #357 (Matrix won), how we obtain the rust-embedded channel, when we move over, what bots or other moderation tools need to be set up.

tim-seoss, 2019-06-24:
Not sure I'll be able to make all (or any) of the meeting, so...

I had a chat on #matrix:matrix.org (and also briefly with Matrix founder Matthew Hodgson) about migrating to matrix.  

The options were:

1. Use the matrix.org home server.  In its favour this is quick and easy (and I've already registered one there).  Downside is that the main matrix.org server is a bit overloaded and until they switch the server software that it's running from the original first-gen Python implementation (Synapse) to the Go rewrite (dendrite) it's likely to remain that way (not until next year).
1. Run a home server e.g. `#embedded-rust:embedded-rust.org` - this would require something like a $10 to $20 per month VPS, plus someone to admin it.  I could set this up if you like, I maintain a few tens of Debian boxes for myself and various clients.   I'm assuming that the WG has control of embedded-rust.org 
1. Get a managed service at http://modular.im/ for $ see https://modular.im/services/matrix-hosting#fair-billing-policy.
1. Use another existing open home server like one of those listed at https://www.hello-matrix.net/public_servers.php or https://matrixservers.net/servers/

It's possible to move later.

Independently of that, there's a question of whether to bridge to IRC (which the WG may or may not wish to do).  If yes, then I think it'd probably be best to initially bridge to the Mozilla IRC channel, and later switch to the freenode IRC channel once the Moz IRC servers go away.

Ralith, 2019-06-26:
A core benefit of matrix is that rooms are not strongly associated with any particular server. Users on any (federated) server can join the room, and the room exists equally on the servers of all participating users. This means that there's no need to (or even any possible way to) commit the community to any particular server; individual users can pick whatever server suits them, and can create new accounts on other servers in the future if desired.

tim-seoss, 2019-06-27:
Yes, apologies, this was my misapprehension.  I had assumed that the room name (more properly "address") or perhaps some metadata would be tied to the namespace where it was created at least in some way.  This isn't the case, and what is now #rust-embedded:matrix.org could also later get an additional address of #main:rust-embedded.org (assuming such a home server was set up) or anything else.

i.e. The room addresses are analogous to DNS records - there can be multiple addresses pointing to the same room, although this is only meaningful if the rooms are federated between servers.

If I have it right, then home server admins get control over their namespace (e.g. which users exist there, and what channels can get aliases etc.), in addition to any performance benefits etc.

nastevens, 2019-07-01:
Sorry I was not able to attend this meeting, it would have been good for me to be there.

Right now provisioning of rust-embedded resources is handled in https://github.com/rust-embedded/rust-embedded-provisioning and hosted on AWS. I pay the bill on that one and I'm totally fine adding more infrastructure as needed within reasonable costs.

If it's determined that we'd like to self-host a home server and/or bridges, that is definitely a possibility. I could either take this on or help mentor someone who would like to do the work.

gbip, 2019-07-29:
Would it be possible to publish the meeting logs somewhere please ? :smile:

stappersg, 2020-02-27:
> Would it be possible to publish the meeting logs somewhere please ?

And then closing this issue?
Or close it now. `8^)`

Clearing stating for which use this issues is still open is also fine.

Plus stating the obvious:  in four months we are past another Jun 25th.

My point:  Open issues  do drain human energy

closed 2020-02-27 by adamgreig