rust-embedded/wg #364: Using LLVM C Backend?

sapir, 2019-07-05:
Would it be possible to use the [LLVM C Backend](/JuliaComputing/llvm-cbe) to generate code for architectures that currently don't have LLVM support, like Xtensa? I've made some PRs there to try to get it to compile rustc's LLVM IR output, so I think it's pretty close to working, at least partially. Is there some reason this wouldn't work? And if it did, how would I add a correct target definition to rustc?

tarcieri, 2019-07-05:
FYI, there's an LLVM Xtensa (WIP) backend: https://github.com/espressif/llvm-xtensa

sapir, 2019-07-13:
Inline assembly doesn't work, because the `asm!` macro parses the assembly as belonging to the current target.

jannic, 2024-06-11:
By now, Xtensa support is working quite well (even if still out-of-tree).

As there was no progress on this ticket for years, I suggest closing it as part of 2024 triage.

MabezDev, 2024-06-11:
Patches are slowly being upstreamed for Xtensa too. The current LLVM main branch can even generate Xtensa machine code (it's not yet usable in Rust though).

Closing as LLVM C backend is unlikely to be the answer here.

closed 2024-06-11 by MabezDev