rust-embedded/wg #365: the embedded rust book inconsistency

josephpenafiel, 2019-07-06:
I couldn't find any site where I could write the Rust guys directly, so here I am. I just want to point out an inconsistency in this book. If you introduce a specific hardware in this book (the STM32F3 discovery board), why are you writing example code about different hardware??? (tm4c123x, in Section 2 Memory map registers). It's just plain confusing and annoying. 
Stick with st or ti, don't mix things up. I think that just scares people away. I know that this book is aimed at people that have some knowledge about embedded systems in general, but my point is about consistency.
That's just something I wanted to get off my chest. Anyone else agree with me? am I wrong?

Regards!

thejpster, 2019-07-06:
That's a useful observation, thank you. Could you open an issue on rust-embedded/book, so we can associate a PR with it?

laanwj, 2019-07-07:
Thanks for reporting !
Moved here: https://github.com/rust-embedded/book/issues/195

closed 2019-07-07 by laanwj