rust-embedded/wg #381: All Hands 2020: Agenda items

jamesmunns, 2019-10-15:
Placeholder

japaric, 2019-10-15:
For reference, this year (RAH 2019) we had 2 1-hour meetings and one 2-hour working session. Notes and attendance can be found at https://paper.dropbox.com/doc/WG-Embedded-Rust-All-Hands-2019-Topics--Ams~zlHKeJ911vGs6mM1Vn~YAg-JrPzUM1Z3SC859MS03lfC

Dylan-DPC, 2019-10-16:
Following this issue for coordination purposes.

jonas-schievink, 2020-01-27:
My two cents:

`T-lang` inquiries:

* Specify what `Send` and `Sync` mean without using the word "thread"
  * Embedded Rust uses "thread-equivalent" constructs such as interrupt handlers, and also has less equivalent constructs such as multi-core MCUs. It would be good to know what `Send` and `Sync` *really* mean, since their contract can't be specific to just threads.
* Reconsider [RFC 2492](https://github.com/rust-lang/rfcs/pull/2492) ("Existential types with external definition") as it would allow using the same mechanism for allocators, panic handlers, and other embedded-specific things (which would be?)

`T-compiler` inquiries:

* How do we get the bug fixed where LLVM thrashes the link register? https://github.com/rust-embedded/cortex-m-rt/issues/139
  * I think there's still no upstream bug for this, so this is basically our fault

`T-infra` inquiries:

* How much room is there on dist-various-1 for embedded targets? Currently that builder runs all embedded tests even though it's not really meant for that I think. (failures also don't show up in PR builds, only in bors)

jamesmunns, 2020-05-26:
All hands didn't happen this year. Closing.

closed 2020-05-26 by jamesmunns