rust-embedded/wg #382: Providing guidance for MSRVs

thejpster, 2019-10-15:
At the EWG meeting 2019-10-15, we discussed providing guidance on how to use MSRVs. This should probably be added to https://github.com/rust-embedded/wg/blob/master/ops/msrv.md.

The guidance should be around:

* What authors of HAL crates should do (the current guidance is only for WG crates)
* How to check what your MSRV is
* Some key features (such as `core::convert::Infallible`) and their MSRV

Steps:

* [ ] Open a PR
* [ ] Get approval
* [ ] Merge the PR
* [ ] Profit

MarcoIeni, 2020-12-26:
I don't know if this the right place to ask for this, but it would be nice to have guidance for MSRV in Github actions
[here](https://github.com/rust-embedded/wg/blob/master/ops/post-transfer.md#ci-via-github-actions-gha).
For example the `1.35.0` is only mentioned [here](https://github.com/rust-embedded/wg/blob/master/ops/post-transfer.md#add-a-githubborstoml-file-with-these-contents) and not in other parts of the document.

BartMassey, 2024-06-11:
Closing this as part of 2024 triage. MSRV issues were resolved by #523.

closed 2024-06-11 by BartMassey