rust-embedded/wg #402: Ecosystem (crater) testing for core crates

ryankurte, 2019-12-03:
given we have a set of foundational crates depended on by practically the whole ecosystem (like https://github.com/rust-embedded/embedded-hal/), the risk of changes to these foundational crates causing large volumes of annoyance is _pretty high_.

it'd be awesome to have https://github.com/rust-lang/crater (or something like it) running over consumers of these crates so we can be informed of and mitigate the impacts of any necessary breakage.

possibly related to: https://github.com/rust-embedded/wg/issues/381

cc. @therealprof

jamesmunns, 2019-12-03:
CC @pietroalbini

pietroalbini, 2019-12-09:
Would you like just cross-compilation or also running tests?

ryankurte, 2019-12-10:
i think just cross compilation (including the test suite), afaik the issues we're looking to avoid are all around types and interfaces so should be p much (where things _can_ compile for a given target) target independent, and shouldn't require test running.

pietroalbini, 2019-12-10:
That should be *possible*, but I'm not sure how it's best to integrate it with Crater off the top of my head.