rust-embedded/wg #403: Agenda for 2019-12-10 meeting

japaric, 2019-12-03:
[8-9 PM CET][0] (Berlin time) on [#rust-embedded:matrix.org](https://matrix.to/#/#rust-embedded:matrix.org)

[0]: https://everytimezone.com/?t=5deee000,474

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

## Agenda 

- The big picture of embedded-hal / re-organizing embedded-hal. cc @thejpster 

- Motivation for book monorepo proposal -- @ghost (who proposed this?)

- Digital v3 (voting? idk) -- @Disasm (?)

- [your topic goes here]

---

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

jamesmunns, 2019-12-04:
Current CI testing is a No-Op: https://github.com/rust-lang/rust/issues/67018

thejpster, 2019-12-06:
Some guidance on https://github.com/rust-embedded/cortex-m-quickstart/pull/78 would be useful. How do we feel about IDE specific additions to quickstart? Would we take an Intelli-J IDEA project, or a Sublime Text project?

rubberduck203, 2019-12-06:
Some context for that conversion @thejpster 

[image]

https://www.jetbrains.com/lp/devecosystem-2019/rust/

jonas-schievink, 2019-12-06:
That PR adds configuration for very specific MCUs, for use with a specific VSCode plugin, not all VSCode users will benefit from that.

Another agenda item: https://github.com/rust-embedded/cortex-m-rt/issues/223 (regression in the cortex-m-rt release).

rubberduck203, 2019-12-06:
@jonas-schievink I didn’t intend to imply that it would. Only providing some editor usage data to frame the conversation.

IGBC, 2019-12-06:
Can we chuck Generic Radio Traits on the agenda?
We have a few driver crates for various radio standards, but when building a protocal ontop of those and making them driver agnostic it would be nice to have a trait for each radio standard (LoRa, Zigbee, Bluetooth, WiFi, whatever) to bind to instead.

therealprof, 2019-12-06:
> Can we chuck Generic Radio Traits on the agenda?

You can put pretty much anything on the agenda but unless you have a proposal it's not very likely that this will result in fruitful discussion since these topics require a high level of domain specific knowledge. It would probably be better to form a group exploring the options and the way forward leading to an RFC proposing those traits and then put it on the agenda for discussion.

IGBC, 2019-12-06:
welp. I will not claim to be experienced in the area.

ryankurte, 2019-12-10:
> Can we chuck Generic Radio Traits on the agenda?

@IGBC i've been working on this for a while now for ISM / LoRa stuff, not sure about how it'd work with WiFi which (afaik) it'd be unusual to interact with as a radio (rather than via some kind of socket API).

You might be interested in [previous discussion](https://github.com/rust-embedded/embedded-hal/issues/48) in embedded-has, some discussion on [rubble](https://github.com/jonas-schievink/rubble/issues/48), and my WIP [implementation](https://github.com/ryankurte/rust-radio) (which links out to a set of implementers). I haven't looked at extracting commonality for configuration yet, but the operation of all of em is the same.

If you'd like to talk about it more we could re-open the HAL issue as a place to chat.

Disasm, 2019-12-10:
Sorry, I will not be able to attend the meeting. As for the `digital::v3`, I'd rather postponed the voting until we agree on the future structure of the `embedded-hal` crate.

thejpster, 2019-12-10:
I'm afraid I've been off sick most of the last week, so I haven't written up the RFC yet.

commit b4731ff1 (2019-12-10): Merge #407

407: Add minutes r=adamgreig a=jamesmunns

Disregard the bad branch name.

closes #403
PR #407 (2019-12-10): Add minutes
Disregard the bad branch name.

closes #403
closed 2019-12-10 by bors[bot]