rust-embedded/wg #405: Missing __truncdfsf2 for aarch64-unknown-none-softfloat

rahealy, 2019-12-04:
I created this issue: [Missing __truncdfsf2 for aarch64-unknown-none-softfloat](https://github.com/rust-lang/compiler-builtins/issues/327) on the "rust-lang/compiler-builtins" repository.

This might be better addressed here since I'm working on an embedded target. Guidance on this matter is definitely appreciated. Should I close the issue in the "compiler-builtins" repo and leave this here? 

Thank you,

Richard

thejpster, 2019-12-04:
While it is an embedded target, we can't add functions to that repository, so you probably do need to raise it with them. If they have any questions we could rummage around for an aarch64 expert, but I'm sure they're pretty hot on the specifics of each microarchitecture anyway.

rahealy, 2019-12-05:
I'll close this issue then. Thank you for the guidance!

closed 2019-12-05 by rahealy