rust-embedded/wg #408: Actual RTFM examples for bluepill?

arrowcircle, 2019-12-16:
Hi! Thanks for the great work bringing rust to embedded devices.
I am kinda  new in Rust and maybe will ask something stupid, but I think it may be helpfull.

1. As I understand, most of the embedded development depends on some kind of RTOS and we have RTFM here. But problem with RTFM, that is does not have actual examples for the most common and cheap hardware, like STM32F3DISCOVERY (as in Rust embedded book) or super cheap $1.5 STM32F103 Bluepill boards. Blog entries from the internet are outdated and does not work. Does it look like a good idea to add working examples for the most common hardware?

2. Rust embedded development relies heavily on alot of crates, but all of this crates does not have proper readme on the github. There are a lot of information in the internet about having great readme page for the open source project and one of the points is to have **working** code examples. Code example makes it easier to start using something. But most of the core embedded crates do not have code examples and moreover, it's very hard to understand what crate do what (cortex-m, cortex-m-rt). There are no link to some central repo, that is easily understandable and easy to start using. Does it make sense to add code examples to core repos and add links to core repo in auxiliary ones?

I think these two points make a huge gap for Rust to be used by more and more people.
Any ideas about that?

jonas-schievink, 2019-12-16:
You don't need to use any RTOS to use embedded Rust, and most crates don't depend on one either. RTFM is used essentially the same way regardless of MCU (you just give it an MCU-specific crate).

I agree about the readme part, we lack docs in general. Feel free to improve this though, most maintainers are pretty responsive.

arrowcircle, 2019-12-16:
> You don't need to use any RTOS to use embedded Rust, and most crates don't depend on one either. RTFM is used essentially the same way regardless of MCU (you just give it an MCU-specific crate).
>
> I [...]

RTFM is not a requirement, but using RTOS makes life much easier. Writing something more complex, than led blink, will be much easier and cleaner with RTOS.

korken89, 2019-12-28:
Hi, thanks for the input!

If you have examples, please make a PR here: https://github.com/rtfm-rs/examples
We are looking to add as many examples as we can for as many boards and MCUs as possible.

arrowcircle, 2020-01-26:
We added example of simple blinky app [https://github.com/rtfm-rs/examples/tree/bluepill/rtfm_v5/bluepill_blinky](https://github.com/rtfm-rs/examples/tree/bluepill/rtfm_v5/bluepill_blinky) (not yet merged)

closed 2020-01-26 by arrowcircle