rust-embedded/wg #411: Transfer ownership of example driver

ryankurte, 2020-01-06:
Hey all,

We talked in other threads about having an example driver / driver template repository to help people navigate the nuances of driver-traits etc.

I've thrown one together [here](https://github.com/ryankurte/rust-embedded-driver), and I think it'd be good to review/expand, transfer to the WG, and to link out from our docs?

almindor, 2020-01-06:
Looking over the code I think it might be more beneficial to write a simple "how-to" blog with these examples instead.

Right now going through it, even tho I wrote a driver myself (only a basic one tho) there were pieces where I went "what? oh.. right" because the example chose an arbitrary interface combination.

A how-to/blog post with maybe a very primitive actual driver as an example but with good step by step explanations would IMO be much more beneficial.

markhildreth, 2020-03-06:
I just started playing with writing a driver myself, and I found the example very useful. I'm not much of a blog reader, but I wonder if the how-to/blog that @almindor mentioned could be part of the README. I blog post could be extra just for those SEO points.

I do agree that some of the parts were confusing for a "general". For example, I had to think for a second if `Error::ResetTimeout` is some weird problem I've never heard of but would be useful to include in my driver, but I believe it's just something that is specific to how this driver works. Perhaps that's just on account of my learning pains to realize this.

markhildreth, 2020-03-06:
Also, just realized this:

`use std::marker::PhantomData;`

Is it okay to have `std` in an embedded driver?

ryankurte, 2020-03-06:
Ahh good catch, this should be `core::marker::PhantomData` (if you need it)

ryankurte, 2020-06-12:
Closing in favour of https://github.com/rust-embedded/book/issues/246

closed 2020-06-12 by ryankurte