rust-embedded/wg #413: Agenda for 2020-01-14 meeting

therealprof, 2020-01-09:
[8-9 PM CET][0] (Berlin time) on [#rust-embedded:matrix.org](https://matrix.to/#/#rust-embedded:matrix.org)

[0]: https://everytimezone.com/?t=5deee000,474

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

## Agenda 

* The road to 1.0:
  * Proposal to restructure `svd2rust` generated code: https://github.com/rust-embedded/svd2rust/issues/427
* [your topic goes here]

---

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

jonas-schievink, 2020-01-09:
We can also discuss https://github.com/rust-embedded/cortex-m/issues/186 and https://github.com/rust-embedded/cortex-m-rt/issues/237 if we have time (they're also part of the road to 1.0).

jonas-schievink, 2020-01-10:
Speaking of changing the svd2rust generated code, I was thinking of changing the API of the register writers so that they make use of traits instead of endlessly chaining fields and values as method calls. So something like `.field().set_bit().field2().variant(Field2::Yes)` would become `.field(true).field2(Field2::Yes)`.

This makes the written code look somewhat cleaner IMO, but the real motivation is unblocking https://github.com/rust-embedded/wg/pull/289. Rustfmt makes the method-chaining-based API look very unreadable since it puts each method call in its own line (https://github.com/rust-embedded/wg/pull/289#issuecomment-452841162).

therealprof, 2020-01-10:
@jonas-schievink Updated accordingly. Feel free to update the issue yourself.

thejpster, 2020-01-10:
Although of course you don't have to chain them.

```
w.field().set_bit();
w.field().baud(19200);
```

I agree that `w.field(true)` is nicer though. Maybe adding `w.set_field(true)` is more backwards compatible.

almindor, 2020-01-10:
I might be wrong but isn't `w.set_bit()` constant (compile time) while `w.set_bit(true)` makes it a runtime expense?

My understanding was that that's the reason why things like `Iterator::rev()` don't have a param.

jonas-schievink, 2020-01-10:
No, both are entirely up to the optimizer. svd2rust marks all of those small functions `#[inline(always)]` to help with that, but that is also just a (strong) hint to the compiler. In practice both should always be optimized out though.

hannobraun, 2020-01-13:
> Speaking of changing the svd2rust generated code, I was thinking of changing the API of the register writers so that they make use of traits instead of endlessly chaining fields and values as method [...]

I don't currently have a strong opinion, but I'd like to note that the proposed style requires the user to import the parameter types, which can be a hassle.

jonas-schievink, 2020-01-13:
@hannobraun That's a good point actually! I think it does match other Rust builder patterns well though.

There might also be a tradeoff regarding compile times here. Do the methods compile faster, or the trait-based approach?

almindor, 2020-01-13:
The more I think about this the more I think we should keep it as-is. Changing functional well understood code to workaround a formatter issue seems backwards.

closed 2020-01-22 by jonas-schievink