rust-embedded/wg #418: Specify the ABI of `-rt` crates in the Rust embedded ecosystem

jonas-schievink, 2020-01-22:
`-rt` crates like [`cortex-m-rt`](https://github.com/rust-embedded/cortex-m-rt/) provide linker scripts with sections and common symbol names that can be used by applications and tooling.

In order for such tooling to be portable across architectures it would be great to specify what kinds of sections and symbols should be exported. This would allow tooling such as [flip-lld](https://github.com/japaric/flip-lld) to work on any architecture (as long as it uses ELF). It can also benefit analysis tools, which can be important on embedded systems.

thejpster, 2020-01-23:
Sounds like a good idea. Should we do our specifications as RFCs like the IETF does, or should we have some sort of `specifications/spec-0000.md` structure do you think?

jonas-schievink, 2020-01-23:
Going through the RFC process seems generally reasonable. One issue might be discoverability when specifications end up among other more "mundane" RFCs. Maybe we want to have an index of specifications in the readme, or something like that?