rust-embedded/wg #422: Agenda for 2020-02-04 meeting

rubberduck203, 2020-02-03:
[8-9 PM CET][timezone] (Berlin time) on [#rust-embedded:matrix.org][matrix]

[Public Google calendar that includes these weekly meetings][google-cal]
Agenda

Leftover from last week:
 
* https://github.com/rust-embedded/cortex-m-rt/issues/237
* digital::v3 RFC voting? https://github.com/rust-embedded/wg/pull/393

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

[timezone]: https://everytimezone.com/?t=5deee000,474
[matrix]: https://matrix.to/#/#rust-embedded:matrix.org
[google-cal]: https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ

rubberduck203, 2020-02-03:
- `r0` v1.0 progress
  - https://github.com/rust-embedded/r0/issues/9
- `bare-metal` v1.0 progress
  - https://github.com/rust-embedded/bare-metal/issues/22

jonas-schievink, 2020-02-05:
Minutes added in https://github.com/rust-embedded/wg/pull/423

closed 2020-02-05 by jonas-schievink