rust-embedded/wg #433: Agenda for 2020-03-17 meeting

jonas-schievink, 2020-03-13:
[8-9 PM CET][timezone] (Berlin time) on [#rust-embedded:matrix.org][matrix]

[Public Google calendar that includes these weekly meetings][google-cal]

Agenda
---

* Default setting for frame pointers, uwtables and backtraces (in light of https://github.com/rust-lang/rust/pull/69890)

Announcements:
* (your announcement here)

---

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

[timezone]: https://everytimezone.com/?t=5deee000,474
[matrix]: https://matrix.to/#/#rust-embedded:matrix.org
[google-cal]: https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ

lenary, 2020-03-15:
I’ll try to join so we can talk about this backtrace issue.

thejpster, 2020-03-17:
Can we talk about Branch management please. Some of our 1.0-readyness changes are breaking, and putting them on master complicates hotfixing existing releases. We need to decide if we want a stable master with feature branches, or an unstable master with release branches. Personally, I'm still a really big fan of "git flow" but I'm aware it's not to everyone's tastes.

thejpster, 2020-03-17:
Also AVR support :tada: :tada: :heart: :+1:

jamesmunns, 2020-03-24:
This meeting happened! Notes will be uploaded today.

closed 2020-03-24 by jamesmunns