rust-embedded/wg #434: Tracking Issue - Team Update/Reorganization

jamesmunns, 2020-03-24:
In the meeting on 2020-03-24 (and in previous meetings), we discussed the following items:

1. we have a number of members which are inactive, which sometimes makes voting difficult
2. we have a number of people who are very active, but don't currently have a role in the WG, but are interested in assisting
3. We have some potential new teams, and maybe some that make sense to retire or rescope
4. We could do better in having an on-ramp for people to help, both on teams, as well as crates in the ecosystem at large. Both in documentation as well as mentoring.

In general, there seems to be interest in reorganizing the teams, and looking at the number of active members on each team.

We will have a meeting on **2020-03-24** focused on the structure of the team. If you have thoughts on the above topics, please make sure to make them known on this issue, or be present at the meeting that day.

We will be putting together a proposal which can then be put through the RFC process.

jamesmunns, 2020-03-24:
CC @rust-embedded/all!

andre-richter, 2020-03-24:
Thanks James.

Most cortex-a team members are effectively dormant/idle.

I am active with respect to the Raspberrypi OS tutorials (mostly) and with register-rs (when maintenance is due, e.g. the Tock project on which it depends changes something on their end).

For the cortex-a crate itself, I see it is being used by people, but I seldom find time to work on it nowadays apart from moderating community PRs.

For one, I could envision removing the cortex-a team and find other homes for said projects.
The tutorials are a natural fit for resources.
register-rs can make sense wherever stuff like svd2rust or “bare-metal” ends up.
The cortex-a crate maybe as well. 

In the past we briefly discussed if having a ”microprocessor” team makes more sense than distinct teams for architectures of specific vendors. If this is to be revisited then cortex-a can find a home there.

Just some thoughts. Depends a lot on where the reorg is going.

nastevens, 2020-03-24:
I can't speak for the rest of @rust-embedded/embedded-linux, but I've been lax on joining weekly meetings since most of the embedded linux work seems to be pretty orthogonal to bare-metal embedded work. I can certainly make a more concerted effort to attend if it would be valuable, although my professional embedded work will remain OS-bound.

jonathanpallant, 2020-03-25:
Could someone post here a list of teams and members, so we can see where we are at the moment?

therealprof, 2020-03-25:
@jonathanpallant Not sure how to extract this information from https://github.com/orgs/rust-embedded/teams . There might be an API to do that but that requires setting up access tokens and might end up a bit annoying.

jonathanpallant, 2020-03-25:
You can't see that if you are not in a team it seems (this is my work account).

therealprof, 2020-03-25:
This should be up-to-date: https://github.com/rust-lang/team/blob/master/teams/wg-embedded.toml
(and is also visible through the Rust homepage). I just noticed that three people are still absent, will add them now. ;)

therealprof, 2020-03-25:
re teams, this is what we have now:

```
All
All members
posborne adamgreig laanwj danc86 raw-bin dvc94ch ryankurte korken89 …
31 members	0 teams 

Core
The core team represents the WG in meetings with the Rust teams
jamesmunns therealprof japaric
3 members	0 teams 

core-team
Core Team of Developers
posborne nastevens rust-embedded-bot
3 members	0 teams 

Cortex-A
Develops and maintains the core of the Cortex-A crate ecosystem
raw-bin andre-richter parched wizofe
4 members	0 teams 

Cortex-M
Develops and maintains the core of the Cortex-M crate ecosystem
adamgreig korken89 thejpster jonas-schievink therealprof ithinuel
6 members	0 teams 

Cortex-R
Develops and maintains the core of the Cortex-R crate ecosystem
japaric paoloteti
2 members	0 teams 

Embedded Linux
Develops and maintains the core of the embedded Linux crate ecosystem
posborne ryankurte nastevens
3 members	0 teams 

HAL
Develops and maintains crates that ease the development of Hardware Abstracti...
ryankurte thejpster therealprof ithinuel
4 members	0 teams 

Highfive
The highfive bot
rust-highfive
1 member	0 teams 

Infrastructure
This teams manages our domains, DNS records, e-mails aliases, etc.
ryankurte nastevens
2 members	0 teams 

IRR 2018
Increasing Rust Reach project 2018 edition
jamesmunns sekineh v-thakkar
3 members	0 teams 

MSP430
Develops and maintains the core of the MSP430 crate ecosystem
pftbest cr1901 awygle
3 members	0 teams 

Resources
Develops, maintains and curates resources on embedded Rust
adamgreig ryankurte korken89 thejpster jamesmunns therealprof andre-richter
7 members	0 teams 

RISCV
Develops and maintains the core of the RISCV crate ecosystem
laanwj danc86 dvc94ch Disasm bradjc almindor
6 members	0 teams 

Tools
Develops and mantains core embedded tools
adamgreig ryankurte reitermarkus Disasm Emilgardis burrbull therealprof
7 members	0 teams 

Triage
The triage team is charge of keeping the PR queue moving
mathk Disasm
2 members	0 teams 
```

paoloteti, 2020-03-27:
Due to changes in company rules I cannot use working hours for not approved open-source projects and I don't have enougth personal spare time to hack on cortex-r devices.
Probably others are in the same situation. A list of members without vote rigths I guess can solve the issue.

dylanmckay, 2020-03-30:
I've been on here since the start, but I only really deal with AVR stuff, of which there is very little on embedded-wg.

That may change with https://github.com/avr-rust/rust/issues/161 however.

I suspect my presence as some kind of admin of this repository may be getting in the way - I don't really interact with much on the embedded-wg because it isn't AVR related. I have no problems with my access being removed to make the process easier - I am not really a voting member.

raw-bin, 2020-03-30:
Hi. 

I don't have a very strong leg to stand on for this request given my mostly peripheral association with the cortex-a team (commentary not code) but I would request that the team be kept alive for 2 reasons:

1. I am on a trajectory with Arm now where my tangible participation through the cortex-a team (and possibly others) will increase. I can provide detail but read on first please.

2. By having a cortex-a focused team, it makes it easier for me to articulate cortex-a specific Rust needs back in the mothership.

Perhaps it might be an idea to simply consider looking at the existing cortex-a team list, checking to see if there has been any commits or commentary and on that basis simply asking the set of folks on that list if they a) Would like to remain members b) Provide a short blurb on how they feel they can provide value moving forwards and then move on that basis ?

On my part, the areas that I can see myself contributing to directly moving forwards:

1. Work with @andre-richter to extend the cortex-a crate with support for architected AArch64 MMUs and Interrupt controllers.

2. Create a port of @phil-opp 's awesome work specific to Arm - by making it a client of 1 above.

Thanks.

jamesmunns, 2025-04-08:
Closing this as probably done, defininitely superceded by #832

cshein45, 2025-04-09:
[Movement](https://github.com/movementlabsxyz)