rust-embedded/wg #442: Move groups and mailinglists to rust team infrastructure

ryankurte, 2020-04-10:
It was recently suggested that we move github group management and email lists to rust team infrastructure, this would be a clear improvement over the current setup (manual management and mailgun respectively).

**Discussions:**
- https://github.com/orgs/rust-embedded/teams/infrastructure/discussions/1
- https://github.com/orgs/rust-embedded/teams/infrastructure/discussions/2

**Questions:**
- What is involved in synchronizing rust-embedded team membership from the teams repo?
- Should we sync _just the overall WG (and resources and core teams) _ or _all teams_ (we have quite a few teams and are likely to have more in the future)?
- Currently our team addresses are `@teams.rust-embedded.org`, are we okay to keep this split?
- Are there any missing actions?

**Actions:**
- Team memberships
  - [ ] Add users and team memberships to https://github.com/rust-lang/team (see WIP)
  - [ ] Update WG team membership documentation / link out to source of truth
- Mailing lists
  - [ ] Add MX records for @rust-embedded.org mailing list hosting
  - [ ] Setup mailing list addresses (?)
  - [ ] Update mailing list addresses / documentation

[WIP](https://github.com/rust-lang/team/compare/master...ryankurte:embedded-teams?expand=1) update to https://github.com/rust-lang/teams

**CC:**
- @pietroalbini as I think you're the person to talk to about this? 
- @nastevens as the other member of @rust-embedded/infrastructure for approval
- @jamesmunns fyi

pietroalbini, 2020-04-10:
> What is involved in synchronizing rust-embedded team membership from the teams repo?

You need to add this to the team repo files of the teams you want to synchronize:

```toml
[github]
orgs = ["rust-embedded"]
```

That snippet will create or synchronize the `@rust-embedded/wg-embedded-infrastructure` team if you enable synchronization for the `wg-embedded-infrastructure` team. You can also specify a different team name with:

```toml
[github]
team-name = "infrastructure"
orgs = ["rust-embedded"]
```

The second snippet will create or synchronize the `@rust-embedded/infrastructure` team.

Also, please `r?` me on the team repo PR.

> Should we sync _just the overall WG (and resources and core teams) _ or all teams (we have quite a few teams and are likely to have more in the future)?

It doesn't make a difference for us how many teams you're synchronizing. I'd lean torwards starting to synchronize a few teams, and if everything works well start synchronizing all the others.

nastevens, 2020-04-18:
This makes total sense to me. It looks like the MX records were provisioned in https://github.com/rust-embedded/rust-embedded-provisioning/pull/13 but that PR was never merged. Let me know if you'd like me to make the MX updates or just include me on the PR.