rust-embedded/wg #445: Tracking issue for MSRV project

adamgreig, 2020-04-14:
This is a tracking issue for progress on https://github.com/rust-embedded/wg/pull/443.

Here's a link to the [project document](https://github.com/rust-embedded/wg/blob/master/projects/in-progress/0443-test-msrv.md) which explains the project in more detail.

# How To Help

We're trying to make sure all WG crates have a well defined Minimum Supported Rust Version (MSRV), which is the oldest version of Rust that can build the crate and run its tests. All our crates should have their MSRV defined in their documentation, and have their CI system automatically run the tests using the MSRV to make sure we don't accidentally break it.

## Getting Started

Thank you for your interest in helping! Depending on your skill level, you might find it easiest to just jump in, or you might want more assistance getting going. If you want to just jump in, pick an unclaimed repository from the list below and follow the steps. You can post here that you're working on a project, or just go ahead and open a PR if it's fairly quick.

If you'd like more assistance, please either leave a comment here, or message @adamgreig:matrix.org on the Rust Embedded Matrix, and we'll get you started!

### 1. Check Current Status

It might be that your repository already has its MSRV documented and/or tested, so it's worth reviewing the steps below quickly to see if this is the case. For repositories where the MSRV is documented _and_ tested, we can tick them off this list with no further work. Please leave a comment here if that's the case!

Otherwise, go ahead and fork the repository from GitHub.

### 2. Finding the MSRV

First, make sure you can build the crate and run the tests. For _most_ WG repositories, that means running their `ci/script.sh` script:

```
$ ci/script.sh
```

Many of our crates build for multiple targets, which is usually defined in their `.travis.yml` CI configuration file. To run your build for a different target, for example `thumbv6m-none-eabi`:

```
$ env TARGET=thumbv6m-none-eabi ci/script.sh
```

Once the build works on your default Rust toolchain, you'll need to test it on other toolchains. To install a custom toolchain (here 1.40.0), a target for it, and then run tests with it:

```
$ rustup toolchain install 1.40.0
$ rustup target add thumbv6m-none-eabi --toolchain 1.40.0
$ env TARGET=thumbv6m-none-eabi RUSTUP_TOOLCHAIN=stable ci/script.sh
```

Once you have this working, the game is to keep reducing the Rust version until it stops working, at which point you've found the MSRV. It might stop working for a trivial reason that is easily fixed, in which case it could be worth fixing and lowering the MSRV further, but we'll have to decide such issues on a case-by-case basis.

On the other hand, if this doesn't work for some reason (some projects have more complicated CI setups), please leave a comment here or open an issue on the relevant repository and we'll help figure out what to do.

### 3. Documenting the MSRV

Once the MSRV is found, add the following snippet to the project's `src/lib.rs` or equivalent documentation:

```rust
//! # Minimum Supported Rust Version (MSRV)
//! 
//! This crate is guaranteed to compile on stable Rust 1.31.0 and up. It *might*
//! compile with older versions but that may change in any new patch release.
```

### 4. Testing the MSRV

To have the CI system run the tests on the MSRV, add the following snippet to the repository's `.travis.yml` configuration file. Note that the `matrix` section probably already exists; you want to add a new entry to it for the MSRV test.

```yaml
matrix:
  include:
    # MSRV
    - env: TARGET=thumbv7-none-eabi
      rust: 1.31.0
    # ...
```

Many projects test on multiple targets, so add an MSRV entry for each target. If in doubt, seek advice on this issue or on the specific project.

### 5. Finishing Up

Make a GitHub PR with your changes and link it here. We'll use the PR to review and discuss the change and then get your changes merged in.

# Crates Checklist

We'll update this list with a user who's working on that repository, a link to the relevant PR when opened, and tick off crates that are complete.

* [ ] [alloc-cortex-m](https://github.com/rust-embedded/alloc-cortex-m)
* [ ] [arm-dcc](https://github.com/rust-embedded/arm-dcc)
* [ ] [bare-metal](https://github.com/rust-embedded/bare-metal)
* [ ] [cargo-binutils](https://github.com/rust-embedded/cargo-binutils)
* [ ] [cortex-a](https://github.com/rust-embedded/cortex-a)
* [ ] [cortex-m-rt](https://github.com/rust-embedded/cortex-m-rt)
* [ ] [cortex-m-semihosting](https://github.com/rust-embedded/cortex-m-semihosting)
* [ ] [cortex-m](https://github.com/rust-embedded/cortex-m)
* [ ] [cortex-r](https://github.com/rust-embedded/patterns)
* [ ] [cross](https://github.com/rust-embedded/cross)
* [ ] [embedded-hal](https://github.com/rust-embedded/embedded-hal)
* [ ] [fixedvec-rs](https://github.com/rust-embedded/fixedvec-rs)
* [ ] [gpio-cdev](https://github.com/rust-embedded/gpio-cdev)
* [ ] [gpio-utils](https://github.com/rust-embedded/gpio-utils)
* [ ] [itm](https://github.com/rust-embedded/itm)
* [ ] [linux-embedded-hal](https://github.com/rust-embedded/linux-embedded-hal)
* [ ] [msp430-rt](https://github.com/rust-embedded/msp430-rt)
* [ ] [msp430](https://github.com/rust-embedded/msp430)
* [ ] [mutex-trait](https://github.com/rust-embedded/mutex-trait)
* [ ] [panic-itm](https://github.com/rust-embedded/panic-itm)
* [ ] [panic-semihosting](https://github.com/rust-embedded/panic-semihosting)
* [ ] [r0](https://github.com/rust-embedded/r0)
* [ ] [register-rs](https://github.com/rust-embedded/register-rs)
* [ ] [riscv-rt](https://github.com/rust-embedded/riscv-rt)
* [ ] [riscv](https://github.com/rust-embedded/riscv)
* [ ] [rust-i2cdev](https://github.com/rust-embedded/rust-i2cdev)
* [ ] [rust-spidev](https://github.com/rust-embedded/rust-spidev)
* [ ] [rust-sysfs-gpio](https://github.com/rust-embedded/rust-sysfs-gpio)
* [ ] [rust-sysfs-pwm](https://github.com/rust-embedded/rust-sysfs-pwm)
* [ ] [svd2rust](https://github.com/rust-embedded/svd2rust)
* [ ] [svd](https://github.com/rust-embedded/svd)

ra-kete, 2020-04-14:
I'd like to help out! I'll take the embedded-hal crate if nobody minds.

ra-kete, 2020-04-14:
Here are the others that already have a MSRV:
- rust-spidev (1.31)
- rust-sysfs-gpio (1.31)
- rust-sysfs-pwm (1.28)
- svd2rust (1.37)

ericevenchick, 2020-04-14:
Put something together to automate this, here are my results so far
```
repos/alloc-cortex-m: MSRV = nightly
repos/arm-dcc: MSRV = 1.31.0
repos/bare-metal: MSRV = 1.31.0
repos/cargo-binutils: MSRV = 1.32.0
repos/cortex-a: MSRV = nightly
repos/cortex-m: MSRV = 1.31.0
repos/cortex-m-rt: MSRV = 1.31.0
repos/cortex-m-semihosting: MSRV = 1.32.0
repos/cortex-r: MSRV = 1.31.0
repos/cross: MSRV = 1.32.0
repos/embedded-hal: MSRV = 1.31.0
repos/fixedvec-rs: MSRV = 1.30.0
repos/gpio-cdev: MSRV = 1.31.0
repos/gpio-utils: MSRV = 1.30.0
repos/itm: MSRV = 1.32.0
repos/linux-embedded-hal: MSRV = 1.32.0
repos/msp430: MSRV = nightly
repos/msp430-rt: MSRV = nightly
repos/mutex-trait: MSRV = 1.31.0
repos/panic-itm: MSRV = 1.31.0
repos/panic-semihosting: MSRV = 1.32.0
repos/r0: MSRV = 1.31.0
repos/register-rs: MSRV = nightly
repos/riscv: MSRV = 1.30.0
repos/riscv-rt: MSRV = 1.31.0
repos/rust-i2cdev: MSRV = 1.31.0
repos/rust-spidev: MSRV = 1.31.0
repos/rust-sysfs-gpio: MSRV = 1.31.0
repos/rust-sysfs-pwm: MSRV = 1.30.0
repos/svd: MSRV = 1.37.0
repos/svd2rust: MSRV = 1.36.0
```

All of the "nightly" MSRVs are due to `#![feature]` being used on the stable channel.
```
error[E0554]: `#![feature]` may not be used on the stable release channel
```

I have only tested from 1.30.0 to 1.42.0.

adamgreig, 2020-04-14:
> Here are the others that already have a MSRV:
> 
> * rust-spidev (1.31)
> * rust-sysfs-gpio (1.31)
> * rust-sysfs-pwm (1.28)
> * svd2rust (1.37)

Thanks for checking! I noticed most of those (and embedded-hal) specify the MSRV in their README instead of in the crate documentation, but I think that's just as effective. svd2rust itself doesn't actually say "MSRV" anywhere, though the README does mention it briefly. It might be worth a quick PR to just include the standard wording.

> Put something together to automate this, here are my results so far

Amazing, thanks!

ra-kete, 2020-04-18:
>  I noticed most of those (and embedded-hal) specify the MSRV in their README instead of in the crate documentation, but I think that's just as effective.

True. Would it make sense to update the "Documenting the MSRV" to mention the README instead of lib.rs, so it gets done consistently? Or alternatively, should we update those crates to list the MSRV at both places?

nastevens, 2020-04-18:
I think it would make sense to incorporate it in both - the README is front and center on [crates.io](crates.io), but the top-level rustdoc is what appears on [docs.rs](docs.rs).

ra-kete, 2020-04-18:
On the other hand it appears to be common practice in Rust projects to document the MSRV in the README only. Personally I wouldn't expect it to be documented on docs.rs, that is more like a reference to me and I usually have gone through the README on GitHub or crates.io before getting there. That doesn't mean putting it into the crate module docs would hurt, although there is some added maintenance burden of one more place to be updated when the MSRV is increased.

ra-kete, 2020-04-19:
[cortex-m](https://github.com/rust-embedded/cortex-m) also already documents and tests its MSRV and can be checked. Though in this case, the MSRV is documented only in the rustdoc, not the README.

adamgreig, 2020-04-19:
I think historically a lot of rust-embedded crates had very minimal READMEs with pretty much just a link to the documentation; personally I'm all in favour of more fully-fledged READMEs alongside the docs. As @nastevens points out, the README is the first thing a user sees on crates.io and on GitHub, while the documentation is what you see once you're on docs.rs, so I think it makes perfect sense to have the MSRV in the README first and foremost.

It is a bit of a pain to have the MSRV duplicated in the README and documentation as well as the CI config file, but maybe worth it for having the MSRV clear in both locations. Anyone else have thoughts on the matter? I'm happy to update the relevant docs to say the MSRV should be documented in the README/both places.

adamgreig, 2020-04-20:
I've updated the list of issues with links to PRs and ticked off those that were merged. I've also opened a PR on cortex-m to change the README there; putting the MSRV in at least the readme seems a good plan. Thanks for all your work so far @ra-kete!

thejpster, 2020-04-21:
Here's my hot take. Careful not to burn yourselves.

I don't care about old versions of stable Rust. The compiler is free. The compiler moves fast, on a regular cadence. Updating is relatively painless. I posit that anyone who *does* care about old versions of stable Rust is in a commercial environment, in which case, I invite them to a) `cargo vendor` everything and b) pay someone to backport fixes. We should not operate like Red Hat Enterprise Linux, as we don't get the subscription fees. We should operate like Debian Testing. Using a rustc compiler from last year is like using Windows XP - OK, but you really need to unplug that think from the Internet, because the rest of us have moved on already.

So what I would do is:

1. define "stable - X", to mean X minor releases prior to the current stable version of rustc. For example, if stable is 1.42, then `stable - 2` is 1.40.

2. Set our MSRV to `stable - 2`. As in the literal string `stable - 2`, not what ever value that happens to evaluate to today. Lazy evaluation if you will.

3. Fix rustup and/or our travis scripts to install `stable - 2`. Whether we grok the version output of rustc and go and `rustup install` one that's two versions older than that, or we patch rustup to understand the concept of `stable - 2` I don't mind.

4. Offer to fix any issues where the crate doesn't build on `stable - 2` but go no further than that.

5. Start using `major.minor.patch` semantic versioning to describe our actual API changes - the bits our users touch on a regular basis - rather than some ill defined concept of maybe 0.001% of our users using last year's compiler are going to have a bad day. If an upstream crate changes something, we only have to worry about the API impact, not whether they now use a feature that was added six months ago. This will require our upstream crates to not be too aggressive in picking up features from the very latest stable, but 12 weeks (i.e. two releases) seems reasonable, as we don't see a huge amount of upstream churn, and it's certainly much less of an issue than setting our MSRV 72 weeks back (i.e. 12 releases) and having to do a bunch of work if anyone starts using a feature stabilised in the last year.

TL;DR: Embrace the release train. Forget about the past. Set the MSRV to `stable - 2` and roll with it.

adamgreig, 2020-04-25:
I've further updated the list with the current PRs and status, thanks everyone!

ra-kete, 2020-04-25:
Mostly all of the crates that have no PR now make use of unstable features and thus require nightly. Those cannot be assigned an MSRV, obviously. But maybe it would be worth documenting that nightly-dependency in the README?

The only exception is [cross](https://github.com/rust-embedded/cross), which has an MSRV of 1.32, except if it invokes xargo, in which case [nightly is required](https://github.com/rust-embedded/cross/issues/154). Not sure how we should deal with that.

adamgreig, 2020-07-07:
Update: All done! Any future crates should ensure they have an MSRV defined, but for now this marks the completion of this focus project. Thanks everyone for all your work getting this done!

closed 2020-07-07 by adamgreig