rust-embedded/wg #457: Books are months out of date when accessed from docs.rust-embedded.org

markhildreth, 2020-05-29:
When accessing any of the three books from the `docs.rust-embedded.org` domain, the version of the documents received are over two months out of date, dating back to March 15th.

```
> date
Fri May 29 15:40:13 EDT 2020
> curl -s --head "https://rust-embedded.github.io/discovery/" | grep last-modified
last-modified: Thu, 28 May 2020 02:46:55 GMT
> curl -s --head "https://docs.rust-embedded.org/discovery/" | grep last-modified
last-modified: Sun, 15 Mar 2020 19:52:23 GMT
> curl -s --head "https://rust-embedded.github.io/book/" | grep last-modified
last-modified: Fri, 29 May 2020 17:35:25 GMT
> curl -s --head "https://docs.rust-embedded.org/book/" | grep last-modified
last-modified: Sun, 15 Mar 2020 19:52:23 GMT
> curl -s --head "https://rust-embedded.github.io/embedonomicon/" | grep last-modified
last-modified: Tue, 21 Apr 2020 17:28:25 GMT
> curl -s --head "https://docs.rust-embedded.org/embedonomicon/" | grep last-modified
last-modified: Sun, 15 Mar 2020 19:52:23 GMT
```

Additionally, the bookshelf is also out of date:

```
> curl -s --head "https://docs.rust-embedded.org" | grep last-modified
last-modified: Sun, 15 Mar 2020 19:52:23 GMT
```

Manually checking, viewing the books through those links do indeed show months-old content. The ci appears to be working, however: the gh-pages branches DO have the updated content, and using the `github.io` domain to access the content provides the updated content as well.

adamgreig, 2020-05-29:
I believe that's because those are all generated/hosted via https://github.com/rust-embedded/docs/tree/gh-pages which has each book as a submodule. Not necessarily a great system...

markhildreth, 2020-05-29:
@adamgreig Ah, didn't realize they were all in that repo. Is there a system in place that should be automating this that isn't working, or is there no such automatic system in place yet?

adamgreig, 2020-05-29:
I'm not sure actually. I don't see anything that would cause a regular build to run, but such things are possible to have on Travis.

yerke, 2020-07-04:
Fixed with https://github.com/rust-embedded/docs/pull/24

closed 2020-07-08 by markhildreth