rust-embedded/wg #459: Embedded register debugging unusable

ryan-summers, 2020-06-21:
When debugging an embedded target over gdb, it is not possible to easily view peripheral register blocks. When printing a peripheral block generated using `svd2rust`, `gdb` will (unhelpfully) display (something similar to) the following:

```
(gdb) print clocks.rb
$1 = stm32h7x::stm32h743v::RCC {_marker: core::marker::PhantomData<*const ()>}
```

This occurs because `stm32h7::stm32h743v::RCC` implements `Deref` to dereference a static address into a `rcc::RegisterBlock` object. However, GDB isn't able to understand the implied dereference and instead just prints the empty RCC container structure.

I've been investigating using a pretty printer in GDB to manually perform the dereference, but it seems that not all the `RegisterBlock` types are available in GDB:
```
(gdb) info types .*RegisterBlock
All types matching regular expression ".*RegisterBlock":

File /home/ryan/.cargo/git/checkouts/stm32h7-ethernet-999804990d5468bc/cf9b8bb/src/lib.rs:
        stm32h7::stm32h743v::ethernet_dma::RegisterBlock;
        stm32h7::stm32h743v::rcc::RegisterBlock;
        stm32h7::stm32h743v::syscfg::RegisterBlock;

File /home/ryan/.cargo/git/checkouts/stm32h7xx-hal-54d30215a9ddb2ce/ff00e93/src/lib.rs:
        stm32h7::stm32h743v::adc3_common::RegisterBlock;
        stm32h7::stm32h743v::flash::RegisterBlock;
        stm32h7::stm32h743v::lptim1::RegisterBlock;
        stm32h7::stm32h743v::lptim3::RegisterBlock;
        stm32h7::stm32h743v::tim1::RegisterBlock;
        stm32h7::stm32h743v::tim2::RegisterBlock;
        stm32h7::stm32h743v::tim3::RegisterBlock;
        stm32h7::stm32h743v::tim4::RegisterBlock;

File /home/ryan/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32h7xx-hal-0.5.0/src/lib.rs:
        stm32h7::stm32h743v::adc3_common::RegisterBlock;
        stm32h7::stm32h743v::flash::RegisterBlock;
        stm32h7::stm32h743v::lptim1::RegisterBlock;
        stm32h7::stm32h743v::lptim3::RegisterBlock;
        stm32h7::stm32h743v::tim1::RegisterBlock;
        stm32h7::stm32h743v::tim2::RegisterBlock;
        stm32h7::stm32h743v::tim3::RegisterBlock;
        stm32h7::stm32h743v::tim4::RegisterBlock;

```

For example, register blocks such as `i2c1::RegisterBlock` and many other peripherals are notable absent.

Additionally, when performing a cast, the output is equally unhelpful/unusable in visualizing the actual contents of the register block:
```
(gdb) print *(0x5802_4400 as &stm32h7::stm32h743v::rcc::RegisterBlock)                                                                                                                                     $5 = stm32h7::stm32h743v::rcc::RegisterBlock {cr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_CR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 251687333}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_CR>}, _reserved_1_icscr: [242, 6, 0,                                                                                                           
    64], crrcr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_CRRCR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 272}}, _marker: core::marker::PhantomData<stm3
2h7::stm32h743v::rcc::_CRRCR>}, csicfgr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_CSICFGR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 536871018}}, _marke
r: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_CSICFGR>}, cfgr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_CFGR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32
> {value: 27}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_CFGR>}, _reserved5: [0, 0, 0, 
    0], d1cfgr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_D1CFGR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 72}}, _marker: core::marker::PhantomData<stm3
2h7::stm32h743v::rcc::_D1CFGR>}, d2cfgr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_D2CFGR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 1088}}, _marker: cor
e::marker::PhantomData<stm32h7::stm32h743v::rcc::_D2CFGR>}, d3cfgr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_D3CFGR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {
value: 64}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_D3CFGR>}, _reserved8: [0, 0, 0, 
    0], pllckselr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_PLLCKSELR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 33686016}}, _marker: core::marker::Phan
tomData<stm32h7::stm32h743v::rcc::_PLLCKSELR>}, pllcfgr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_PLLCFGR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 312
60706}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_PLLCFGR>}, pll1divr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_PLL1DIVR> {register: vcell::VolatileCell<u32> {value: cor
e::cell::UnsafeCell<u32> {value: 16842951}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_PLL1DIVR>}, pll1fracr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_PLL1FRACR> {registe
r: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 0}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_PLL1FRACR>}, pll2divr: stm32h7::generic::Reg<u32, stm32h7::stm32
h743v::rcc::_PLL2DIVR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 16975559}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_PLL2DIVR>}, pll2fracr: stm
32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_PLL2FRACR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 0}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::
rcc::_PLL2FRACR>}, pll3divr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_PLL3DIVR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 16843392}}, _marker: core::mar
ker::PhantomData<stm32h7::stm32h743v::rcc::_PLL3DIVR>}, pll3fracr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_PLL3FRACR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32>
 {value: 0}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_PLL3FRACR>}, _reserved16: [0, 0, 0, 
    0], d1ccipr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_D1CCIPR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 0}}, _marker: core::marker::PhantomData<stm
32h7::stm32h743v::rcc::_D1CCIPR>}, d2ccip1r: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_D2CCIP1R> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 0}}, _marker: 
core::marker::PhantomData<stm32h7::stm32h743v::rcc::_D2CCIP1R>}, d2ccip2r: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_D2CCIP2R> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCe
ll<u32> {value: 0}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_D2CCIP2R>}, d3ccipr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_D3CCIPR> {register: vcell::VolatileCell<u32> 
{value: core::cell::UnsafeCell<u32> {value: 0}}, _marker: core::marker::PhantomData<stm32h7::stm32h743v::rcc::_D3CCIPR>}, _reserved20: [0, 0, 0, 
    0], cier: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_CIER> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 0}}, _marker: core::marker::PhantomData<stm32h7::
stm32h743v::rcc::_CIER>}, cifr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_CIFR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 0}}, _marker: core::marker::Pha
ntomData<stm32h7::stm32h743v::rcc::_CIFR>}, cicr: stm32h7::generic::Reg<u32, stm32h7::stm32h743v::rcc::_CICR> {register: vcell::VolatileCell<u32> {value: core::cell::UnsafeCell<u32> {value: 0}}, _marker:
 core::marker::PhantomData<stm32h7::stm32h743v::rcc::_CICR>}, _reserved23: [0, 0, 0, 

<snipped>
```

The easiest method to visualize the actual register contents is to just examine the raw memory in GDB and cross-reference with a datasheet, which is extremely unappealing and time consuming.

```
(gdb) x/10wx 0x5802_4400
0x58024400:     0x0f0071a5      0x400006f2      0x00000110      0x2000006a
0x58024410:     0x0000001b      0x00000000      0x00000048      0x00000440
0x58024420:     0x00000040      0x00000000
```

adamgreig, 2020-06-21:
Should this issue be in svd2rust rather than wg? It only affects svd2rust-generated crates, and probably would be fixed there if anywhere. Unfortunately I can't see any good way to fix it for svd2rust as-is. The possible future move to not using Deref and just having raw pointers instead of C-layout structs won't help either.

However, I can highly recommend https://github.com/bnahill/PyCortexMDebug which is a gdb extension to use SVD files to aid debugging:

```
>>> svd
Available Peripherals:
        FSMC:    Flexible static memory controller
        PWR:     Power control
        RCC:     Reset and clock control
        GPIOA:   General purpose I/O
<snip>
>>> svd/x I2C1
Registers in I2C1:
	CR1:    0x00000001  Control register 1
	CR2:    0x00000018  Control register 2
	OAR1:   0x00000000  Own address register 1
	OAR2:   0x00000000  Own address register 2
	DR:     0x000000B0  Data register
	SR1:    0x00000001  Status register 1
	SR2:    0x00000003  Status register 2
	CCR:    0x00000078  Clock control register
	TRISE:  0x00000019  TRISE register
>>> svd I2C1 SR1
Fields in SR1 of peripheral I2C1:
	SMBALERT:  0  SMBus alert
	TIMEOUT:   0  Timeout or Tlow error
	PECERR:    0  PEC Error in reception
	OVR:       0  Overrun/Underrun
	AF:        0  Acknowledge failure
	ARLO:      0  Arbitration lost (master mode)
	BERR:      0  Bus error
	TxE:       0  Data register empty (transmitters)
	RxNE:      0  Data register not empty (receivers)
	STOPF:     0  Stop detection (slave mode)
	ADD10:     0  10-bit header sent (Master mode)
	BTF:       0  Byte transfer finished
	ADDR:      0  Address sent (master mode)/matched (slave mode)
	SB:        1  Start bit (Master mode)
```

ryan-summers, 2020-06-21:
I was originally debating placing this in the `svd2rust` crate, but I also realized that I don't think there's a good fix that can be implemented alone there. I think there may be more work throughout the ecosystem needed to improve usability, but I'd be happy to move this if you think it's not appropriate here.

ryan-summers, 2020-06-21:
I believe that https://github.com/rust-lang/rust/issues/33014 may be relevant here for supporting `Deref` within GDB, but I'm not familiar enough with the internals of DWARF and GDB to know for sure.

ryan-summers, 2020-06-22:
I'm now moving this to the above-mentioned issue in `svd2rust` as I agree with @adamgreig that it may be more appropriate there after I have done some further investigation into this problem.

closed 2020-06-22 by ryan-summers