rust-embedded/wg #462: Update "Organization -> Other projects" section of README.md to list well supported devices

bbrown1867, 2020-06-28:
As a new Rust embedded user, I would like to see what devices are "well supported" by the Rust community. By these I mean devices that have a sub-community around them, not a single repo created by an individual. For example, the nRF and STM32 microcontrollers appear to fit this definition:
* https://github.com/nrf-rs
* https://github.com/stm32-rs

My understanding is that these communities are not part of the working group, and the working group focuses on supporting architectures rather than devices (e.g. cortex-m, cortex-r, risc-v, etc.). Please correct me if I'm wrong.

Just linking to these communities from the working group README and stating that they are separate communities would help I think. Happy to make the PR.

Disasm, 2020-06-28:
I'd like to note that STM32 support for different STM32 families varies from mature to "a single repo created by an individual".

bbrown1867, 2020-06-28:
Perhaps that wasn't the best way to describe it. I just meant that the repos under `stm32-rs` and `nrf-rs` appear to have some sort of community wrapped around them, even if they are developed by one person.

hannobraun, 2020-06-29:
There's lpc-rs too: https://github.com/lpc-rs

eldruin, 2020-06-29:
I think more visibility for these high-profile projects would be a good thing.

bbrown1867, 2020-07-02:
Can I open a PR for this issue? Or does it need to be approved by someone first?

therealprof, 2020-07-02:
@bbrown1867 Well, we first need to see the proposed changes before approving them. 😅 

Please feel free to PR your ideas and I'll be happy to look into moving your PR forward.

bbrown1867, 2020-07-02:
Okay great! :) Thought maybe there some issue triage process or something

therealprof, 2020-07-02:
Well, there is a process for more complex proposals which require some exploration and discussion to decide on technical details. But for more lightweight changes like I expect this one to be I'd start with a proposal and then we can look at whether this is simply a matter of someone rubberstamping it (or suggesting a better alternative) or it warrants a few minutes of discussion in the next meeting or it needs a fully fledged RFC and vote.

This kind of sounds like it go in the first category but we'll see when there's something to see. ;)

bbrown1867, 2020-07-08:
Closed by #466

closed 2020-07-08 by bbrown1867