rust-embedded/wg #480: Location for the new DMA traits

thalesfragoso, 2020-07-15:
After some discussion in the DMA focus project, it was decided that we need a new set of traits to completely cover all the necessary requirements for DMA transfers. More explanation and experimentation can be found at this [repo](https://github.com/ra-kete/dma-poc).

Right now we're in the phase of testing the implementations. However, I think we can start discussing a place for these traits to live, because if we decide for a new crate it would be nice to create at least the repo for it in the organization for better visibility, which might help in the gathering of more implementations.

I think the only existing place we could place them would be in `embedded-hal` but I'm not sure if this is the best place for them.

therealprof, 2020-07-21:
CC https://github.com/stm32-rs/stm32f4xx-hal/pull/186

therealprof, 2020-07-21:
As discussed in todays meeting please continue bikeshedding on this issue. People seem to be in favour of a separate repository/crate but no decision yet.

therealprof, 2020-07-28:
Looked at this and #486 in the meeting today. From a brief discussion people still seem to be in favour of a separate crate.

My bikeshed name proposoal: `embedded-dma`

ra-kete, 2020-07-28:
Regarding bike shedding, I'd like to throw just `dma` out. Seems less redundant since DMA is always embedded, no? But then again, we already have the pattern of prefixing crates with `embedded-` so doing that here would help with the brand recognition I guess.

therealprof, 2020-08-04:
@ra-kete No, DMA is definitely not related to embedded alone. Pretty much everything is using DMA in the PC world (even over USB and FireWire which is a funny way to hack a system).

therealprof, 2020-08-04:
Discussion from todays meeting: We're going ahead with the suggested name `embedded-dma`. @thalesfragoso to set it up.

thalesfragoso, 2020-08-11:
The repository for the new crate is up here: https://github.com/rust-embedded/embedded-dma

What is left to discuss is which team will be the responsible for it, looking at the teams descriptions, it seems to me that the best suited for this task is the HAL team. 

I would like the opinions of the HAL team members about this, @rust-embedded/hal . Although I'm not in the HAL team, I will still help in the maintaining of this crate in specific.

therealprof, 2020-08-18:
As discussed today, @thalesfragoso will publish as is and add the @rust-embedded/hal as maintainer.

thalesfragoso, 2020-08-23:
Everything is sorted out now, so closing this.

closed 2020-08-23 by thalesfragoso