rust-embedded/wg #482: Rust embedded community logo based on EWG logo

eldruin, 2020-07-20:
We want to add a logo to the [rust-embedded-community](https://github.com/rust-embedded-community).
We liked a logo based on the EWG logo so it would be appropriate to ask here for approval.
Here is the proposed logo:
[image]
[Discussion issue](https://github.com/rust-embedded-community/meta/issues/3)
Opinions? Concerns?

therealprof, 2020-07-21:
As discussed in the meeting, this should be clarified with the Rust core team and get a written permission regarding the Rust trademark.

therealprof, 2020-08-25:
cc https://github.com/rust-embedded-community/meta/issues/3

eldruin, 2020-09-01:
A summary of the options as requested in the last meeting would be:

- Adopt REC as a child REWG org
    - Membership requirements TBD.
    - Maybe the original logo proposal would be acceptable (needs re-evaluation by core team)
- REC stays as a separate org.
    - Needs alternative logo to original proposal.
        - Ferris-based one may be too similar to REWG's.
    - REC name deemed confusing w.r.t. REWG.
        - New name necessary.
    - REC name deemed fine w.r.t. confusion with REWG.
        - Name stays as-is.

eldruin, 2020-09-07:
I added an initial RFC with the terms for adopting REC as a child org at https://github.com/rust-embedded/wg/pull/504.

eldruin, 2020-11-12:
Closing as the RFC will not be merged.

closed 2020-11-12 by eldruin