rust-embedded/wg #491: Bring embedded-nal under the umbrella

bobmcwhirter, 2020-08-06:
There's at least a non-zero number of interested parties for

https://github.com/thejpster/embedded-nal

I propose we bring it under the embedded umbrella (if @thejpster agrees) to continue development of the traits and publishing of the crate.

ryan-summers, 2020-08-06:
For context, I have interest in using the `embedded-nal` for [`minimq`](https://github.com/quartiq/minimq)

bobmcwhirter, 2020-08-06:
Also note that I’m willing to help be a maintainer even though I prove on a daily basis to have NFI what I’m doing. Willing to try.

ryankurte, 2020-08-06:
if/when @thejpster is ready this seems good to me. related to https://github.com/rust-embedded/wg/issues/348

ryan-summers, 2020-08-07:
Also related to https://github.com/rust-embedded/embedded-hal/issues/146

newAM, 2020-08-09:
[`embedded-nal`](https://github.com/thejpster/embedded-nal) depends on [`no-std-net`](https://github.com/dunmatt/no-std-net), should this be brought under the embedded umbrella as well?

Ideally this dependency will not be needed in the future if [RFC 2832](https://github.com/rust-lang/rfcs/pull/2832) is successful in adding a `core::net`.

therealprof, 2020-08-11:
We briefly discussed this in the meeting today. Unfortunately no one has spoken out on behalf of the proponents of taking this under the embedded WG umbrella.

Personally I'd like to see a vision for further developments (what is missing? why does this require embedded WG supervision instead of being a community driven project?) and people interested in taking on the stewardship (and potentially becoming a WG member).

bobmcwhirter, 2020-08-11:
I understand the reasoning as described in the meeting.

I also grok that "network" is probably over-broad, given the different levels of networking, from interfaces to UDP to TCP etc.

@thejpster -- would you accept PRs against the repo and release crates for experimentation? Should someone fork and start afresh?  I was under the impression that you no longer had time to investigate/push the crate forward, hence my suggestion for the wg.  If there's another route, I'm all ears.

eldruin, 2020-08-11:
The [rust-embedded-community](https://github.com/rust-embedded-community) might also be an alternative for it if maintenance is the problem.

ryan-summers, 2020-08-12:
> We briefly discussed this in the meeting today. Unfortunately no one has spoken out on behalf of the proponents of taking this under the embedded WG umbrella.

I meant to show up for the WG meeting, but had some real-life obligations to attend to instead. Sorry about that!

There's a few points I want to make here. 

> why does this require embedded WG supervision instead of being a community driven project?)

I've just written an MQTT implementation and want to target various network stacks (as pointed out by @adamgreig) such as `smoltcp` and an external W5500 TCP/IP stack chip. One way to do this is to define by own trait that users can choose to implement, or the `embedded-nal` can serve as the abstraction layer.

I originally thought this would be good to take on as part of the embedded-wg to prevent fractionalization (e.g. if every crate is using their own custom traits, it's really not ergonomic for users). If there's a single, well-maintained trait, if a network stack implements that trait, it can immediately be used without extra work from the end user.

This could also work as a community driven project, but I fear that crate maintainers may see it as too "unofficial" to adopt. How should a maintainer react when a different implementation arrives and there's two competing crates? It could become overwhelming from a maintenance perspective.

> what is missing?  

There's definitely room for growth. The current API forces a slice copy, which prevents zero-copy network abstractions. If we could expand it, we can then support high-performance, zero-copy network abstractions fro higher level protocols. There's also definitely more protocols that we could potentially expand to (such as potentially raw IP packets or something else, although this is out of my area of expertise).

> people interested in taking on the stewardship

I'd be happy to help with maintence as well as (I believe) @bobmcwhirter 

> @thejpster -- would you accept PRs against the repo and release crates for experimentation? Should someone fork and start afresh? I was under the impression that you no longer had time to investigate/push the crate forward, hence my suggestion for the wg. If there's another route, I'm all ears.

I believe @thejpster is currently on hiatus, which has essentially stalled development of the `embedded-nal`. Right now, I don't think it's possible to incorporate changes into the original crate.

adamgreig, 2020-09-01:
Closing as embedded-nal has now been moved to rust-embedded-community :tada:

closed 2020-09-01 by adamgreig