rust-embedded/wg #529: AVR flashing tool. Help needed

luqasz, 2020-12-14:
Hi. 

I've started working on some [code](https://github.com/luqasz/flash) for flashing  / reading AVR atmega chips. As for now I got working reading flash from atmega32. atmega2560 seems to be a bit odd. I'd like to contribute to ecosystem as native rust flashing tool will be a good tool / library.

If any one has some time to help then please contact me either via repo issue or mail.

ryan-summers, 2020-12-15:
Have you checked out [`probe-rs`](https://github.com/probe-rs/probe-rs) yet? I know they target mainly ARM cores, but I think it would be well-worth checking out the work there and potentially contributing with them.

I would also highly recommend the [Rust Embedded Matrix Channel](https://matrix.to/#/#rust-embedded:matrix.org) for discussions like this

luqasz, 2020-12-15:
Ok. Thx. I will continue with matrix and probe-rs

rursprung, 2024-06-19:
not sure if this issue is still relevant? [avr-hal](https://github.com/Rahix/avr-hal) (which provides a HAL for the ATmega and a few other AVR chips) has a guide on flashing, it comes with its own tool, [ravedude](https://github.com/Rahix/avr-hal/tree/main/ravedude#readme) which internally uses [avrdude](https://github.com/avrdudes/avrdude/). this does however mean, that it's (sadly) not integrated in [probe-rs](https://probe.rs/), so you still need that separate installation of the tool.

luqasz, 2024-06-19:
Hi. 
I've developed a bit of code to flash, read/write duses into avr devices with stk500 programmer. https://crates.io/crates/avrisp Feel free to use it. 
I'd like to develop it further but I have other duties now, so can't promise anything. You can hit me tough if you want some information on how things work.

rursprung, 2024-06-19:
CC @rahix (since you built `ravedude` & `avr-hal`), see above