rust-embedded/wg #539: Mailing list?

alfiedotwtf, 2021-02-10:
Sorry... I tried searching everywhere but couldn't find anything. Is there a place where I can sign up to a WG mailing list? Also, the newsletters too.

Thanks 🎉

therealprof, 2021-02-10:
👋

we don't run a mailinglist but we'd be happy to see you on our Matrix/Element.io channel https://app.element.io/#/room/#rust-embedded:matrix.org

Regarding the newsletter, you may want to point your favourite RSS reader to https://rust-embedded.github.io/blog/rss.xml to receive updates as they are published.

alfiedotwtf, 2021-02-10:
Thanks for the quick response!

I'm more of a lurker, so Matrix is probably too fast-moving for me right now (the Discovery board sits on my desk covered in dust... guilt-tripping me), but I'll definitely follow the feed :)

FrankenApps, 2021-04-28:
Whenever I try to join the matrix room, I get a "Room alias not found" error. Maybe this can be looked into.

ryankurte, 2021-04-28:
via [element.io](https://app.element.io/#/room/#rust-embedded:matrix.org)? it's all working as far as i can see

FrankenApps, 2021-04-29:
I retried using the Web version of element and now it worked. Never had this problem using the mobile app though, so maybe it's some sort of configuration Issue?

therealprof, 2021-04-29:
No idea, works just fine for me.