rust-embedded/wg #611: Calliope Mini

Darneas, 2022-04-05:
Dear all,
(Hope this is the right spot for this)

Recently, I got my hands on a Calliope Mini (a board used in Germany to help elementary school students explore computing). I was wondering if there are existing efforts to bring Embedded Rust to this target. What I saw is that the chip already has core support (cortex M0) and basic peripherals (nRF51822). However, a proper board support package is missing. Additionally, proper support by probe-rs / probe-run should be checked.

Are there others who would be interested in bringing Rust to the Calliope Mini?

Thanks in advance!

Best,
Andreas

therealprof, 2022-04-05:
You might want to check out https://github.com/nrf-rs/microbit which has quite a few similarities to the Calliope.

jamesmunns, 2024-06-11:
Closing this as part of the 2024 triage for inactivity, @netzdoktor feel free to swing by the Matrix room to discuss if this is still relevant!

closed 2024-06-11 by jamesmunns