rust-embedded/wg #647: Tracking Issue: Rust Embedded on Mastodon

jamesmunns, 2022-11-22:
Following up from chat, we probably want to have some kind of presence on Mastodon/fediverse.

The two main open questions:

**Who is going to run it?**

IMO (@jamesmunns): whoever runs it should be on the Resources team. We should also figure out how to share access to the account to avoid it getting abandoned, etc.

* @mygnu has volunteered

We may want an existing RE member, maybe @thejpster or @adamgreig, to help mentor whoever runs the account.

@jamesmunns can help with "open source brand" style and coming up with general practices to be consistent with the twitter and cohost accounts.

**Which server should it be on?**

There are open discussions on this. Chat discussed:

* hachyderm.io
* fosstodon.org

mygnu, 2022-11-24:
I'm happy to help also in favour of hachyderm.io (biased cos I already have an account there) I believe @thejpster is also there. We should definitely have some guidelines around how to operate such account.

thejpster, 2022-11-25:
I'm @thejpster@hachyderm.io.

I understant Kris at Hachyderm.io, like many volunteer instance mods, has a *lot* going on right now. So whoever we approach we should do it with care and understanding.

thejpster, 2022-12-04:
Hachyderm has had a big infra upgrade. Could be a good home now the dust has settled.

adamgreig, 2022-12-06:
Sounds good to me! I've set up @rustembedded@hachyderm.io, and am just sorting out having it go to the resources team email address. As far as I could tell there's no reason not to just create the account their; registrations are open. I don't think there's any way to share ownership beyond just sharing the password, which I'm happy to do.

We should work out some basic principles for operating the accounts as @jamesmunns suggested; perhaps something like:

1. Post about specific announcements from the working group, e.g. to tell people about some major new release, a breaking change, etc
2. Post WG blog posts
3. Repost/share people offering and seeking jobs, with some rate limit that doesn't feel too spammy and will have to adjust based on how many requests we get
4. Repost/share people's posts tagged/mentioning us about embedded rust, same deal with the rate limit
5. Repost/share other major Rust related posts, e.g. from the official Rust account

How does that sound?

mygnu, 2022-12-06:
Sounds good to me

thejpster, 2022-12-07:
yup, good starter for ten.

jamesmunns, 2022-12-07:
Opened https://github.com/rust-embedded/wg/pull/651, feel free to add any feedback or suggestions there.