rust-embedded/wg #669: Benchmarking embedded rust code

KRACK-BIT, 2023-04-05:
I've been working on my own embedded rust project (a no_std KYBER implementation), and have been trying to run time consistency tests on my micro-controller. As such, I'm currently trying to validate that the number of clock cycles are constant for each of my critical functions, and I can measure the clock cycle time directly through cortex_m::Peripherals

However, I can't find any bench-marking framework which supports measuring this, or even any projects that take measurements (be they clock or real-world time) of a micro-controller's runtime.

As such, are there any good libraries/examples on profiling execution time on a micro-controller, and if not, are there any examples on how to write your own bench-marking framework for an embedded system?

ryan-summers, 2023-04-05:
For cortex-m devices, the ITM can be used to do this if your microcontroller can support it. [probe-rs](https://github.com/probe-rs/probe-rs) supports this functionality, but it's not a ready-made solution, so there would be a non trivial implementation to do this. Take a look at [rtic-scope](https://github.com/rtic-scope) if you're using RTIC

therealprof, 2023-04-14:
An approach has been mentioned in https://blog.japaric.io/rtfm-overhead/. I've been also using the external method using a GPIO quite successfully to do A/B comparisons of code and optimize it, though it won't yield exact measurements.

There have been some experiments with https://github.com/orbcode/orbuculum a long time ago but I can't find any reference. Using that software with ITM/ETM has the potential of yielding professional level results and details at the expense of a non-trivial setup and potentially research work.