rust-embedded/wg #710: Transfer heapless and volatile-register to libs team

adamgreig, 2023-10-17:
I propose (with @japaric's blessing!) transferring [heapless](https://github.com/japaric/heapless) and [volatile-register](https://github.com/japaric/volatile-register) repos to the WG under the libs team. Heapless is widely used across the embedded ecosystem but the current maintainers don't have much time for updates, while volatile-register is used by cortex-m and we'd like to at least fix a soundness issue for us and other users (https://github.com/rust-embedded/cortex-m/issues/485). The newly formed libs team is a perfect fit, not coincidentally.

I hope we'll have a few more members soon, now that the team exists. Still, the current libs team should vote:

- [ ] @Dirbaio

Dirbaio, 2023-10-17:
I vote yes :laughing:

newAM, 2023-10-17:
I vote yes!  ♥️ heapless

japaric, 2023-10-19:
> I propose (with @japaric's blessing!) transferring [heapless](https://github.com/japaric/heapless) and [volatile-register](https://github.com/japaric/volatile-register) repos to the WG under the libs team

I'm on board :+1: (let me know when / where to transfer the repos)

Dirbaio, 2023-10-19:
I don't think there's any blocker left? you can transfer to https://github.com/rust-embedded, @adamgreig will have to configure the permissions I guess.

adamgreig, 2023-10-19:
> I'm on board 👍 (let me know when / where to transfer the repos)

Thanks! Just transfer them to rust-embedded any time, I'll then assign them to the libs team.

japaric, 2023-10-20:
done!

- https://github.com/rust-embedded/heapless
- https://github.com/rust-embedded/volatile-register

adamgreig, 2023-10-20:
Thank you! I've put them under the libs team now. Could you add the team on crates.io as well, please?

```
cargo owner --add github:rust-embedded:libs
```

adamgreig, 2023-10-20:
@rust-embedded/libs, there's some notes in https://github.com/rust-embedded/wg/blob/master/ops/post-transfer.md about the changes to make, which for this I think boil down to:

* [ ] update wg/README.md with new repos
* [ ] update repo READMEs with reference to the libs team and CoC
* [ ] add CoC file to repo
* [ ] update repo `.github/CODEOWNERS` with `* rust-embedded/libs`
* [ ] possibly set up GHMQ/required CI checks if desired

japaric, 2023-10-20:
@adamgreig it seems like I need to be temporarily added to the `libs` team to do that:

``` console
$ cargo owner --add github:rust-embedded:libs
error: failed to invite owners to crate `heapless` on registry at https://crates.io

Caused by:
  the remote server responded with an error: only members of a team or organization owners can add it as an owner
```

adamgreig, 2023-10-20:
Huh, I didn't know that! I wonder if it's a recent change. Could you add me as an owner directly instead, then? At least that also avoids not being able to update owners in future, and I should be able to add the libs team.

japaric, 2023-10-20:
@adamgreig crates.io invites sent!

adamgreig, 2023-10-20:
Thanks, all sorted!

Dirbaio, 2023-10-20:
all done! thanks guys :partying_face:

closed 2023-10-20 by Dirbaio