rust-embedded/wg #720: Tracking Issue: Call for blog posts, EoY 2023

jamesmunns, 2023-11-28:
It has been roughly 5 years since "stable embedded Rust", and it seems like a reasonable excuse to solicit blog posts from folks during the month of December.

In particular, it would be interesting to hear blog posts on the following topics:

## The Good

I'd love to see blog posts that explore any/all of:

* Positive stories you have about Embedded Rust and/or the Embedded Rust community, including personal or professional experiences
* Particular crates, ecosystems, tools, or workflows that you like or have found useful
* Thanks for people who have helped you, or you'd like to show appreciation for

I think it's worthwhile to take some time to talk positively about things, share good experiences, and show and tell a little bit, as so often people are missing good things that they just don't know exist, or don't even know what to ask for.

## The Bad

I'd also be interested in hearing about:

* Things that were, or are still challenging
* Pain points (solved or unsolved) over the past years
* Things that you wish were better, or are still missing

In particular, this is NOT to set up a roadmap, and DEFINITELY not to put anyone or any project on blast. My hope for this is to clearly outline issues you've personally faced, in a way that helps other people understand the issues, and serve as a starting point for future discussons on how things could be resolved.

Criticism is fine, but please be professional and remember that most people here are volunteering their time and efforts.

## The Other

If you just want to write, to be heard, or to finally get that blog post out in the world, and want to use this as the reason, then go for it!

## Then what

For now, if you write a blog post, please link it here as a comment on this issue. At the end of December/early January, @jamesmunns will do a "wrapup" overview, and promote the posts as much as possible.

## NOTE

This isn't "official" yet, if you have thoughts on "the process", feel free to chime in below!

nixpulvis, 2023-11-29:
I would be very interested in a talk about the current state of cross-compiling and flashing of various hardware. I'm used to both: the raspberry pi method of putting binaries on an SD card and the AVR method of FTDI uploading with a standalone tool (`avrdude`). It would be magnificent if we could unify this workflow if it hasn't already been.

ia0, 2023-12-14:
Not a blog post per se, but it still covers some of the aspects in "The Good", in particular useful crates of the ecosystem: https://google.github.io/wasefire/acknowledgments.html

apollolabsdev, 2023-12-15:
Great idea @jamesmunns. Here's my two-cent:
https://apollolabsblog.hashnode.dev/embedded-rust-education-2023-reflections-2024-visions

apollolabsdev, 2023-12-22:
> > Great idea @jamesmunns. Here's my two-cent: https://apollolabsblog.hashnode.dev/embedded-rust-education-2023-reflections-2024-visions
> 
> Dude, you're awesome - your blog posts have been incredibly helpful in inducting me into a side of rust I believe I never would've been able to enter otherwise. I seriously wish documentation wasn't as scarce as it is - I'm used to the rust where I can easily determine what functions do but now like 40% of the time, functions/structs simply don't have a description

Thank you for your kind words! I'm thrilled to hear that my blog posts have been helpful in your journey into embedded Rust. It's always a pleasure to share knowledge and make the learning process more accessible. If there's anything specific you'd like to see covered in future posts or if you have any questions, feel free to let me know. Maybe I can cover it in a future post. Happy coding!

EdBuilds, 2023-12-23:
Hello, I could give this a try.
I just finished my first blog post here: [test-driven-embedded-rust-development](https://hackaday.io/page/21907-test-driven-embedded-rust-development)
Let me know if I could improve on it.

JuliDi, 2023-12-30:
Thanks, @jamesmunns for the great idea. We'd like to share a blogpost here https://www.systemscape.de/en/blog/2023/wrap-up/ about the great experience we've had so far and a big "thank you" to some of the most helpful folks in the community. At [Systemscape](https://github.com/Systemscape/) we're very convinced of a bright future for embedded rust and are confident that we're doing the right thing, starting our life after university in this field.

jamesmunns, 2024-02-05:
Also should grab the embassy release notes: https://embassy.dev/blog/embassy-hals-released/

jamesmunns, 2024-02-05:
Also should grab the e-hal 1.0 release notes: https://blog.rust-embedded.org/embedded-hal-v1/

bsodmike, 2024-02-05:
I just posted this getting started guide https://desilva.io/posts/debugging-arm-cortex-m-with-jlink-mini-and-arduino-portenta-h7-giga-r1-wifi-board

jonathanpallant, 2024-02-05:
At work we wrote about:

* [Rust on ThreadX](https://ferrous-systems.com/blog/rust-and-threadx/)
* [Rust on SPARC](https://ferrous-systems.com/blog/rust-for-mission-critical-applications/)
* [Using `mem::take` to reduce heap allocations](https://ferrous-systems.com/blog/rustls-borrow-checker-p1/) in rustls