rust-embedded/wg #753: embedded-can appears to be in limbo

reneherrero, 2024-05-06:
Hi,

There's much confusion around the state and direction of embedded-can:

1. The stm32 fdcan project has [commented it out](https://github.com/stm32-rs/fdcan/blob/7955ae21bca3b7556b5d810a3bfae85e3b41a0df/src/lib.rs#L31)
2. According to the stm32 bxcan project, the embedded-can crate is [deprecated](https://github.com/stm32-rs/bxcan/blob/3fc7a0e81975d4f25e61e0da81cd9e7a5e969e81/CHANGELOG.md?plain=1#L20).
3. There's a [mismatch in uniformity](https://github.com/rust-embedded/embedded-hal/issues/599) compared to embedded-hal
4. CAN FD is a pretty big thing as it bumps the payload size from 8 bytes to 64 bytes, but there's [no word](https://github.com/rust-embedded/embedded-hal/issues/578) on that

Can anyone shed some light into the matter?

Thanks

tristanguerragomez, 2024-05-06:
IMHO, this is causing fragmentation in the ecosystem

liamkinne, 2024-07-05:
@reneherrero @tristanguerragomez glad to see others share my concerns about the state of CAN in the ecosystem.

I've just gotten my PR merged (https://github.com/rust-embedded/embedded-hal/pull/612) that adds defmt traits to embedded-can so I think there is appetite from the maintainers for improvement, there just isn't anyone to do it.

I'm hoping to work on CAN FD in the coming weeks which should pave the way for the fdcan crate to properly support the embedded-can traits.

eldruin, 2024-07-05:
Any improvements are definitely welcomed and encouraged. However, this is a volunteer-based project where each person has specific interests.
In the past, there simply has been more interest in the traits included in `embedded-hal`, which is why that is more polished. 
We rely on the whole community to engage in discussions and *especially* to make proposals for improvements and/or to solve problems that arise in the implementations across the ecosystem. 
As I said, please feel encouraged to drive CAN support forward if you are interested in it. You will be most welcome.

projectgus, 2024-10-17:
Raising my hand as interested in contributing to this as well. I've just popped up a draft PR to partially support existing embedded_can types in fdcan driver: https://github.com/stm32-rs/fdcan/pull/56 but I'd be interested in expanding embedded_can further as well. (In particular: CAN FD support as per @liamkinne, and support for split controllers with separate CanRx and CanTx traits).