rust-embedded/wg #754: Actions-rs is deprecated

nastevens, 2024-05-07:
Per https://github.com/actions-rs the actions-rs project is deprecated and unmaintained. The associated actions have started to raise numerous warnings and are likely to stop working entirely at some point. An unscientific survey of other Rust projects seems to indicate a migration to https://github.com/dtolnay/rust-toolchain, but I would like to get some consensus from the team before charging ahead with this. This may also be an opportunity to close out #671.

I'm opening this issue here since it affects at least 20 R-E repos across multiple subteams at the time of writing: https://github.com/search?q=org%3Arust-embedded+actions-rs&type=code.

Rahix, 2024-05-07:
There is also [actions-rust-lang/setup-rust-toolchain](https://github.com/actions-rust-lang/setup-rust-toolchain) but I don't know how the two compare.

newAM, 2024-05-08:
It was also pointed out to me that `rustup` can be used directly, it's included in the ubuntu runner images.

jonathanpallant, 2024-05-08:
I prefer to just use rustup.

nastevens, 2024-05-08:
Does anyone have repos they can point to as reference of good design using GHA? Bonus points if the repo is already testing a matrix of architectures and features.

Emilgardis, 2024-05-08:
svd2rust has a pretty good setup: https://github.com/rust-embedded/svd2rust/blob/6979dc40c2d812b6f4f0b4fc7a18c29233faaf85/.github/workflows/ci.yml#L50

romancardenas, 2024-05-09:
In [`riscv`](https://github.com/rust-embedded/riscv/blob/master/.github/workflows/riscv.yaml) we use `rust-toolchain` with different targets, features, etc.

So far, it works pretty fine.

jannic, 2024-05-14:
I updated several repos to `dtolnay/rust-toolchain` a while ago, eg.:
- https://github.com/rust-embedded/embedded-hal/pull/439
- https://github.com/rp-rs/rp-hal/pull/532

Not saying that this is the best approach, only noting that it seems to work well.