rust-embedded/wg #768: Figure out automation of the "all" team on GH

jamesmunns, 2024-06-19:
We have a team, "all", that is not managed by the rust-lang teams bots.

[image]

All other teams have a toml that has a section like this:

```toml
[[github]]
orgs = ["rust-embedded"]
team-name = "resources"
```

If we want the bot to manage the team access for the "all" account, I think we either need to:

* add a `wg-embedded-all.toml` manifest to the `rust-lang-team/teams` folder: https://github.com/rust-lang/team/tree/master/teams
* Add the `[[github]]` metadata to the `wg-embedded.toml` manifest, treating it as the source for the "all" team.
* Get rid of the `all` team, as it's manually maintained atm

CC @therealprof, who I THINK set this all up originally. I'm not familiar enough with teams automation to know which option is right, but this came up with thejpster retiring.

BartMassey, 2024-06-19:
Option 4: Construct the `all` team as the union of the team manifests in CI. Optionally check to ensure that this matches the `[[github]]` manifest.

Or is the `[[github]]` manifest already this?

jamesmunns, 2024-06-19:
@BartMassey looking at the toml docs: https://github.com/rust-lang/team/blob/master/docs/toml-schema.md

It appears we could instead:

* remove all `people.members` of `wg-embeded.toml`
* add a `people.included-teams` section, listing all substeams
* add the `[[github]]` section as proposed above

jamesmunns, 2024-06-19:
We should also add `website.zulip-stream` while we are at it.