rust-embedded/wg #771: Soundly initializing custom memory sections

ryan-summers, 2024-06-28:
Currently, pointer provenance appears to make it entirely impossible to correctly initialize a memory section within rust code because linkers communicate with the application by setting the address of variables. These passed-back variables then have a provenance that does not cover the entire memory region that needs to be defined. For example:
```rs
extern "C" {
    static mut __sitcm: u32;
    static mut __eitcm: u32;
    static mut __siitcm: u32;
}

unsafe {
    let sitcm = core::ptr::addr_of_mut!(__sitcm); // Provnenance of 4 bytes
    let eitcm = core::ptr::addr_of_mut!(__eitcm); // Provenance of 4 bytes
    let siitcm = core::ptr::addr_of_mut!(__siitcm); // Provenance of 4 bytes
    let len = eitcm.offset_from(sitcm) as usize;

    // This is UB because `sitcm` and `siitcm` only have provenance of 4 bytes from their respective addresses, when `len` > 4.
    core::ptr::copy_nonoverlapping(siitcm, sitcm, len);
}
```

Right now, it doesn't appear possible to soundly perform section initialization using data provided by the linker due to these provenance issues. I'm opening this issue to track the topic, as we were discussing this extensively in the Matrix channel earlier.

jamesmunns, 2024-06-28:
https://github.com/rust-embedded/cortex-m-rt/issues/300 is the discussion where we removed this pattern from `cortex-m-rt`

jamesmunns, 2024-06-28:
https://github.com/rust-lang/unsafe-code-guidelines/issues/259

I hadn't seen this update, it may be that statics are now more magical than I was arguing.