rust-embedded/wg #782: Run 2024 Embedded WG Micro-Survey

jamesmunns, 2024-07-30:
I realized that the Rust survey team has the option to run "micro surveys" on behalf of Rust teams: https://github.com/rust-lang/surveys/blob/main/micro-surveys.md

I wonder if it's worth us putting together a survey for the embedded ecosystem, asking things like:

- Are you currently using embedded rust for hobby/school/work/none/in the past but not now?
- How long have you been using embedded Rust (0-6+ years)
- If using for work, how many devices/year is your company shipping currently using Rust?
- What architectures are you using (Arm/RISC-V/Xtensa/other)
- What chip families are you using (STM32/nRF52/RP/ESP32/etc)
- What frameworks/RTOSs are you using (Bare/Embassy/RTIC/Tock-OS/ESP-IDF/Hubris/other)
- How are you using Rust? (Only Rust, Rust as primary with C components, C as primary with Rust components, Rust tooling only (off device))
- What tools do you use as part of your development process (Cargo, probe-rs, defmt, other)
- What are the most common crates/libraries you use in your projects (free entry)
- What are your biggest complaints or blockers for using Rust in your projects (free entry)?
- How happy are you with using Rust for embedded (1-10)?

Open to feedback and suggestions for questions, note that we should not ask for any PII or DEI information, as per the micro survey guidelines

_Originally posted by @jamesmunns in https://github.com/rust-embedded/wg/discussions/780#discussioncomment-10158216_

jamesmunns, 2024-07-30:
Next steps:

1. [ ] - We need to review the language of the questions (feedback welcome here for now), it is currently only in a rough draft state
2. [ ] - We need to plan the "PR effort" for this - where do we get this included, how much time do we need it open for, who do we need to coordinate with?
3. [ ] - @jamesmunns to open a PR to the /wg repo with the questions in the format required by the survey team
4. [ ] - We submit the PR to the survey team and go live

CodingInGreen, 2024-07-31:
I just want to reference the comment I made about the importance of keeping the person taking the survey engaged enough during the survey that they will complete the survey and click the submit button.

This typically means less questions, simpler questions, and anything we can do to reduce the amount of mental effort someone has to exert while taking the survey.

CodingInGreen, 2024-07-31:
On a separate thread, we mentioned two surveys with different purposes in mind.

@jamesmunns would you mind putting together a list of purposes for the survey. I can think of a few.

1. What is the current state of Rust adoption in Embedded, and trajectory for the future within organizations.
2. Collected data that would be useful for Silicon Vendors, that would further the purposes of Rust adoption in Embedded.

U007D, 2024-08-03:
Hey James, great idea.

It's also important to ask where people *want* to go and what they *want* to do beyond where they are now.  Things like `Error` in `core`, the demand for a fledgling microarchitecture, language changes and so on could all be uncovered with these questions.  Asking participants to identify their top friction point with the language and with the ecosystem could help too.

Thank you!

jamesmunns, 2024-08-13:
@U007D and @CodingInGreen I've opened #787, please let me know if you have any feedback there!

jamesmunns, 2024-08-22:
This survey is now live: https://www.surveyhero.com/c/uenp3ydt

I'll be tracking all of the places that it has been shared in this message. Please feel free to comment any additional places below, so we don't miss anything.

diondokter, 2024-08-22:
I posted on my personal account on mastodon: https://fosstodon.org/@diondokter/113005581427839200

CodingInGreen, 2024-08-22:
Shared in a LinkedIn post!

spookyvision, 2024-08-22:
[shared](https://chaos.social/@dngrs/113005793979254980) on mastodon

michelrandahl, 2024-08-22:
[reddit](https://www.reddit.com/r/rust/comments/1eyjyd7/2024_embedded_wg_microsurvey/)

chrysn, 2024-08-22:
Dropped it in the [RIOT OS matrix chat](https://matrix.to/#/!pqHdpanAvkJvlCwUDE:matrix.org/$NWXLn75xEJ3-dubN3x2mhlKK6XfLqnNqRs3yoGBdyVE?via=matrix.org&via=utwente.io&via=tu-dresden.de)

kevswims, 2024-08-22:
Shared in the Embedded.fm podcast Slack

jamesmunns, 2024-08-22:
Shared in the Interrupt/Memfault Slack

BartMassey, 2024-08-22:
This Week In Rust?

9names, 2024-08-23:
Posted on the following Discord channels:
- The Rust Programming Language: #wg-embedded
- Rust Programming Language Community: #embedded
- Embedded Engineering: #rust
- Pico (Unofficial): # rust

chrysn, 2024-08-23:
> This Week In Rust?

[Like this?](https://github.com/rust-lang/this-week-in-rust/pull/5852)

Dirbaio, 2024-08-23:
shared it in more places
- https://docs.embassy.dev menu bar
- Zephyr discord, #rust channel
- Pigweed discord, #rust channel

BartMassey, 2024-08-23:
@chrysn Thanks much!

alevy, 2024-08-23:
Shared on:

- The Tock developer mailing list: https://lists.tockos.org/hyperkitty/list/devel@lists.tockos.org/thread/HFX4VVGB55M2UVCHDP3SOHVNH4PTJ2DB/
- Tock socials (retweeted and retooted)
- Will post to Tock website header (as @Dirbaio did for embassy) ASAP

sirhcel, 2024-08-31:
[Shared](https://matrix.to/#/!cnJwoUGZmLziNcGNua:matrix.org/$Y8qaOT2BNLIT7tpA27FTq_k0C_i8I_uCfHA1vlitllU?via=matrix.org&via=t2bot.io&via=matrix.shymega.org.uk) it on the [serialport-rs Matrix channel](https://matrix.to/#/#serialport-rs:matrix.org)

jordens, 2024-12-03:
I can't find the survey results or info about where and when they will be posted.
The inside rust blog post or any other posts I could find don't mention anything about the results.
Some survey results were [posted in the channel](https://matrix.to/#/!BHcierreUuwCMxVqOf:matrix.org/$F8d5ayZThh9RECgdAkXzCtjk1v_wInM2vGAcNnr0L8M?via=matrix.org&via=catircservices.org&via=tchncs.de) two months ago. Presumably the full analysis is still out because it needs to be word-clouded/manually written up.

Is this issue here the landing page for people looking for results?
Could we update this issue (and/or any other prominent mentions of the survey, like [the blog post](https://blog.rust-lang.org/inside-rust/2024/08/22/embedded-wg-micro-survey.html)) with info about results availability?
For any subsequent surveys it might be helpful to already put info/links regarding the results in the call for submissions. Otherwise, once the survey closes, the call immediately becomes a high-profile dead end.

jamesmunns, 2024-12-03:
Hey @jordens, that's a fair request. The full tables and result numbers are here: https://github.com/rust-embedded/wg/discussions/796#discussioncomment-10742588

As you mentioned, I haven't had time to review and publish the "raw" text results (there is definitely PII in a few of the responses, we need to take a curation pass on this before we can publish it). Then yes, we will need to figure out how to post these results, but I think that another inside rust post is fine. @BartMassey was the only person who volunteered to assist with this, but I haven't had time to ping him either (and I'm guessing he's been busy and hasn't pinged me either).

Agreed we can do better next year for making sure that the calls include a placeholder link to the eventual destination for results.

BartMassey, 2024-12-03:
I'm still happy to help; ping me when you've got some time and we'll do this. (Quite busy right now, next week would be a bit better. But I'll find time whenever.)