rust-embedded/wg #827: Unable to disable Z extensions when targeting riscv32imac-unknown-none-elf

Surplars, 2025-03-22:
I'm developing an embedded project targeting riscv32imac-unknown-none-elf and need to disable the Z extensions (specifically zmmul, zaamo, and zalrsc). However, even after setting the following rustflags in my Cargo.toml:``
```
[target.riscv32imac-unknown-none-elf]
rustflags = [
    "-C", "target-feature=-zmmul,-zaamo,-zalrsc"
]

```
However, when I use code that iterates over a string (which calls internal library functions for UTF-8 decoding), the build fails with a Z extension error. For example:
```
#[unsafe(no_mangle)]
extern "C" fn main() {
    uinit();

    let str: &str = "Hello, RUST!\n";
    for c in str.chars() {
        uart_put_char(c as u8);
    }

    loop {}
}
```
This results in a linker error like:
`... -march=rv32i2p1_m2p0_a2p1_c2p0_zmmul1p0_zaamo1p0_zalrsc1p0: unsupported ISA subset 'z'`
However, a simpler version of the code compiles, links, and runs correctly:
```
#[unsafe(no_mangle)]
extern "C" fn main() {
    uinit();

    uart_put_char(b'R');

    loop {}
}
```

Rust Version: rustc 1.87.0-nightly (be73c1f46 2025-03-21)
RV32I / M / A / F / D / C / P / B instruction extensions support.

Expected Behavior:
The target should respect the provided rustflags and disable the Z extensions, generating a parameter that only includes supported ISA subsets.-march

Actual Behavior:
The generated parameter still includes , , and , causing the external linker to fail with an unsupported ISA subset error.-marchzmmulzaamozalrsc

Request:
Could you please help ensure that the Z extensions can be fully disabled when using the target? Alternatively, any guidance or documentation on how to properly disable these extensions for this target would be greatly appreciated.riscv32imac-unknown-none-elf

Thank you for your assistance!

romancardenas, 2025-03-22:
HI @Surplars ! Can you share a repo so I can reproduce the issue on my side?

Surplars, 2025-03-22:
> HI [@Surplars](https://github.com/Surplars) ! Can you share a repo so I can reproduce the issue on my side?

Sorry, I tried to reproduce the issue, and surprisingly it linked successfully. This is really strange.

closed 2025-03-22 by Surplars