rust-embedded/wg #828: Transfer ownership of `accelerometer` to libs team?

romancardenas, 2025-03-22:
Hi!

I think it would be a good idea to transfer ownership of the [`accelerometer`](https://github.com/NeoBirth/accelerometer.rs) crate to the libs team. This crate is a nice abstraction for accelerometers, which simplifies significantly the code for retrieving the orientation of a device, for instance.

The authors is currently no maintaining this project, but is willing to transfer the ownership of the project to keep it updated.

What do you think @rust-embedded/libs ?

newAM, 2025-03-22:
I don't have the time to maintain it at the moment.  I'm not even properly keeping up with existing crates under the libs team umbrella :( If somebody else on the libs team has time to maintain it then I'm all for it.

It might be worth starting the discussion in rust-embbeded-community as well.

romancardenas, 2025-03-23:
Sounds good to me! There are a lot of great crates that become unmaintained, and it is a pity. The embedded Rust community may come up with a mechanism to keep them up to date.

I currently only belong to the `riscv` team, but I volunteer myself to help the libs team maintaining crates such as `accelerometer` or similar. Probably I'm not eligible to become a team member, but I would be happy being a collaborator.

reitermarkus, 2025-04-25:
I wouldn't mind as long as we have someone on the libs team actually using the crate. Out of the crates under the libs team, I'm really only actively using `heapless`, so adding another crate that is passively maintained isn't much better than it being unmaintained.

jonathanpallant, 2025-04-27:
Personally I don't think it's foundational enough to be in EDWG.

REC seems a better fit. 

But this opinion is weak and I am open to persuasion.

BartMassey, 2025-04-28:
REC? Sorry, I'm still kind of new here 😀 .

jonathanpallant, 2025-04-29:
Ah, sorry.

GitHub.com/rust-embedded-community

BartMassey, 2025-04-29:
Thanks! Given that this is the first I've heard of REC, we should probably find a more effective way to tell people about it 😀 .

closed 2025-06-23 by romancardenas