rust-embedded/wg #847: Discussion Issue: WG Website and Showcase Review

ryanstreur, 2025-07-01:
- [WG Agenda Item](https://github.com/rust-embedded/wg/discussions/846#discussioncomment-13500700)
- [Matrix Discussion](https://matrix.to/#/!BHcierreUuwCMxVqOf:matrix.org/$KQlNuEra1DOpGAN0OvbXxtni1OnJobAtATCCwRrHWww?via=matrix.org&via=catircservices.org&via=beeper.com)

## Questions:

Partucularly for @nastevens and @ryankurte:  

- Is the [TF Script for provisioning the WG website](https://github.com/rust-embedded/rust-embedded-provisioning) still in use?
- Who is currently paying for the infrastructure?

## Decisions

- If the TF script is in use, should we move to a potentially-cheaper static hosting platform like GitHub pages or similar?
- If the TF script is not in use
    - Do we want to reactivate it as a project?
    - If not, shall we archive or remove it?
- [The Embedded Showcase](https://github.com/rust-embedded/showcase) has not been updated in about four years. Shall we merge it with the [Awesome Embedded Rust repo](https://github.com/rust-embedded/awesome-embedded-rust/) by moving its entries over and deleting it (while keeping the DNS entry to prevent its registration by domain scalpers)?

BartMassey, 2025-07-01:
As long as it's a subdomain of `rust-embedded.org` I don't think there's any domain scalping issue.

BartMassey, 2025-07-01:
I would love to see folks update the Embedded Showcase. I was just looking through it and there's some great stuff there. I don't think any of it has aged out too much.

I think the hard question is how to make it more visible. I suspect that it is not updated because Rust embedded developers largely aren't aware of it.

I'd be fine seeing it move wherever it is most visible, especially if it is easier to maintain.

eldruin, 2025-07-02:
I had a PR to make the showcase the landing page (rust-embedded.org) [here](https://github.com/rust-embedded/showcase/pull/33). Might be worth a revisit.

BartMassey, 2025-07-02:
Do we have instructions somewhere on how to contribute to the Showcase?

eldruin, 2025-07-03:
@BartMassey the [showcase repo](https://github.com/rust-embedded/showcase) itself has instructions, including evaluation criteria.

BartMassey, 2025-07-03:
How did I miss this? Sincere apologies.

I will start recruiting at least one project I know of. I think it would be good to update the showcase.